cesanta/mongoose #911: Cygwin compilation issue (undefined reference, starting 6.7)

benrubson, 2018-04-10:
Hi,

Mongoose 6.7 and 6.11 (not tested between) with Cygwin gives the following compilation errors.
No issue with 6.6, so I think the issue is here since 6.7.

```
$ make
gcc restful_server.c ../../mongoose.c -o restful_server -g -W -Wall -Werror -I../.. -Wno-unused-function  -DMG_ENABLE_THREADS -DMG_ENABLE_HTTP_WEBSOCKET=0 -lws2_32
/tmp/ccFTTvZE.o: In function `ev_handler':
/mongoose-6.11/examples/restful_server/restful_server.c:33: undefined reference to `mg_vcmp'
/mongoose-6.11/examples/restful_server/restful_server.c:35: undefined reference to `mg_vcmp'
/tmp/cc2H1nvc.o: In function `mg_normalize_uri_path':
/mongoose-6.11/examples/restful_server/../../mongoose.c:5274: undefined reference to `mg_vcmp'
/mongoose-6.11/examples/restful_server/../../mongoose.c:5276: undefined reference to `mg_vcmp'
/tmp/cc2H1nvc.o: In function `mg_http_serve_file':
/mongoose-6.11/examples/restful_server/../../mongoose.c:6845: undefined reference to `mg_vcmp'
/tmp/cc2H1nvc.o:/mongoose-6.11/examples/restful_server/../../mongoose.c:7290: more undefined references to `mg_vcmp' follow
collect2: error: ld returned 1 exit status
make: *** [../examples.mk:30: restful_server] Error 1
```

With Cygwin 64, there are additional errors :
```
$ make
gcc restful_server.c ../../mongoose.c -o restful_server -g -W -Wall -Werror -I../.. -Wno-unused-function  -DMG_ENABLE_THREADS -DMG_ENABLE_HTTP_WEBSOCKET=0 -lws2_32
/tmp/ccINLf78.o: In function `ev_handler':
/mongoose-6.11/examples/restful_server/restful_server.c:33: undefined reference to `mg_vcmp'
/mongoose-6.11/examples/restful_server/restful_server.c:33:(.text+0x165): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-6.11/examples/restful_server/restful_server.c:35: undefined reference to `mg_vcmp'
/mongoose-6.11/examples/restful_server/restful_server.c:35:(.text+0x19e): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/ccnapON9.o: In function `mg_normalize_uri_path':
/mongoose-6.11/examples/restful_server/../../mongoose.c:5274: undefined reference to `mg_vcmp'
/mongoose-6.11/examples/restful_server/../../mongoose.c:5274:(.text+0x9902): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-6.11/examples/restful_server/../../mongoose.c:5276: undefined reference to `mg_vcmp'
/mongoose-6.11/examples/restful_server/../../mongoose.c:5276:(.text+0x991d): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/ccnapON9.o: In function `mg_http_serve_file':
/mongoose-6.11/examples/restful_server/../../mongoose.c:6845: undefined reference to `mg_vcmp'
/mongoose-6.11/examples/restful_server/../../mongoose.c:6845:(.text+0xc33c): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/ccnapON9.o:/mongoose-6.11/examples/restful_server/../../mongoose.c:7290: more undefined references to `mg_vcmp' follow
/tmp/ccnapON9.o: In function `mg_check_digest_auth':
/mongoose-6.11/examples/restful_server/../../mongoose.c:7290:(.text+0xdd8b): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-6.11/examples/restful_server/../../mongoose.c:7291:(.text+0xdda6): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/ccnapON9.o: In function `mg_http_send_port_based_redirect':
/mongoose-6.11/examples/restful_server/../../mongoose.c:7540:(.text+0xeaa0): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/ccnapON9.o: In function `mg_uri_to_local_path':
/mongoose-6.11/examples/restful_server/../../mongoose.c:7779:(.text+0xf851): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-6.11/examples/restful_server/../../mongoose.c:7781:(.text+0xf86c): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/ccnapON9.o: In function `mg_is_creation_request':
/mongoose-6.11/examples/restful_server/../../mongoose.c:7905:(.text+0xfe63): additional relocation overflows omitted from the output
collect2: error: ld returned 1 exit status
make: *** [../examples.mk:30: restful_server] Error 1
```

This is the example for the `restful_server`.
The `simplest_web_server` for example compiles correctly.

Thank you 👍 

Ben

Dorokun192, 2019-09-03:
Any solutions for this in mongoose 6.15?

benrubson, 2020-02-07:
Any hint perhaps @cpq ? Many thx 👍

cpq, 2020-02-11:
Please do all tests against HEAD.

Weird error. I don't see it under mingw and native VS build.
Could you confirm that an example builds correctly under mingw please?

benrubson, 2020-02-14:
Thank you for your response @cpq.

Actually, same error with last Cygwin 64 (3.1.2) and last Mongoose commit (https://github.com/cesanta/mongoose/commit/359060a151998c5ab08e54af1420b7406005dc6b) :
```
$ make
gcc restful_server.c ../../mongoose.c -o restful_server -g -W -Wall -Werror -I../.. -Wno-unused-function  -DMG_ENABLE_THREADS -DMG_ENABLE_HTTP_WEBSOCKET=0 -lws2_32
/tmp/cccW4jw7.o : In function `ev_handler':
/mongoose-master/examples/restful_server/restful_server.c:33 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/restful_server.c:33:(.text+0x165): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/restful_server.c:35 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/restful_server.c:35:(.text+0x19e): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_normalize_uri_path':
/mongoose-master/examples/restful_server/../../mongoose.c:5543 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5543:(.text+0xa1c5): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5545 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5545:(.text+0xa1e0): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_get_mime_type_encoding':
/mongoose-master/examples/restful_server/../../mongoose.c:5995 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5995:(.text+0xac1c): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o:/mongoose-master/examples/restful_server/../../mongoose.c:7172 : encore plus de références indéfinies suivent vers « mg_vcmp »
/tmp/cclYIxpM.o : In function `mg_http_serve_file_internal':
/mongoose-master/examples/restful_server/../../mongoose.c:7172:(.text+0xcf3a): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_check_digest_auth':
/mongoose-master/examples/restful_server/../../mongoose.c:7626:(.text+0xeb38): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:7627:(.text+0xeb53): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_http_send_port_based_redirect':
/mongoose-master/examples/restful_server/../../mongoose.c:7875:(.text+0xf84a): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_uri_to_local_path':
/mongoose-master/examples/restful_server/../../mongoose.c:8114:(.text+0x105f8): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:8116:(.text+0x10613): débordement de relocalisation additionnelle omise à partir de la sortie
collect2: error: ld returned 1 exit status
make: *** [../examples.mk:30: restful_server] Error 1
```

Thx !

benrubson, 2020-02-14:
Thank you for your response @cpq.

Actually, same error with last Cygwin 64 (3.1.2) and last Mongoose commit (https://github.com/cesanta/mongoose/commit/359060a151998c5ab08e54af1420b7406005dc6b) :
```
$ make
gcc restful_server.c ../../mongoose.c -o restful_server -g -W -Wall -Werror -I../.. -Wno-unused-function  -DMG_ENABLE_THREADS -DMG_ENABLE_HTTP_WEBSOCKET=0 -lws2_32
/tmp/cccW4jw7.o : In function `ev_handler':
/mongoose-master/examples/restful_server/restful_server.c:33 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/restful_server.c:33:(.text+0x165): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/restful_server.c:35 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/restful_server.c:35:(.text+0x19e): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_normalize_uri_path':
/mongoose-master/examples/restful_server/../../mongoose.c:5543 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5543:(.text+0xa1c5): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5545 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5545:(.text+0xa1e0): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_get_mime_type_encoding':
/mongoose-master/examples/restful_server/../../mongoose.c:5995 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5995:(.text+0xac1c): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o:/mongoose-master/examples/restful_server/../../mongoose.c:7172 : more undefined references to `mg_vcmp' follow
/tmp/cclYIxpM.o : In function `mg_http_serve_file_internal':
/mongoose-master/examples/restful_server/../../mongoose.c:7172:(.text+0xcf3a): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_check_digest_auth':
/mongoose-master/examples/restful_server/../../mongoose.c:7626:(.text+0xeb38): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:7627:(.text+0xeb53): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_http_send_port_based_redirect':
/mongoose-master/examples/restful_server/../../mongoose.c:7875:(.text+0xf84a): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_uri_to_local_path':
/mongoose-master/examples/restful_server/../../mongoose.c:8114:(.text+0x105f8): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:8116:(.text+0x10613): débordement de relocalisation additionnelle omise à partir de la sortie
collect2: error: ld returned 1 exit status
make: *** [../examples.mk:30: restful_server] Error 1
```

Thx !

benrubson, 2020-02-14:
Thank you for your response @cpq.

Actually, same error with last Cygwin 64 (3.1.2) and last Mongoose commit (https://github.com/cesanta/mongoose/commit/359060a151998c5ab08e54af1420b7406005dc6b) :
```
$ make
gcc restful_server.c ../../mongoose.c -o restful_server -g -W -Wall -Werror -I../.. -Wno-unused-function  -DMG_ENABLE_THREADS -DMG_ENABLE_HTTP_WEBSOCKET=0 -lws2_32
/tmp/cccW4jw7.o : In function `ev_handler':
/mongoose-master/examples/restful_server/restful_server.c:33 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/restful_server.c:33:(.text+0x165): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/restful_server.c:35 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/restful_server.c:35:(.text+0x19e): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_normalize_uri_path':
/mongoose-master/examples/restful_server/../../mongoose.c:5543 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5543:(.text+0xa1c5): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5545 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5545:(.text+0xa1e0): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_get_mime_type_encoding':
/mongoose-master/examples/restful_server/../../mongoose.c:5995 : undefined reference to `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:5995:(.text+0xac1c): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o:/mongoose-master/examples/restful_server/../../mongoose.c:7172 : more undefined references to `mg_vcmp' follow
/tmp/cclYIxpM.o : In function `mg_http_serve_file_internal':
/mongoose-master/examples/restful_server/../../mongoose.c:7172:(.text+0xcf3a): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_check_digest_auth':
/mongoose-master/examples/restful_server/../../mongoose.c:7626:(.text+0xeb38): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:7627:(.text+0xeb53): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_http_send_port_based_redirect':
/mongoose-master/examples/restful_server/../../mongoose.c:7875:(.text+0xf84a): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/tmp/cclYIxpM.o : In function `mg_uri_to_local_path':
/mongoose-master/examples/restful_server/../../mongoose.c:8114:(.text+0x105f8): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `mg_vcmp'
/mongoose-master/examples/restful_server/../../mongoose.c:8116:(.text+0x10613): additional relocation overflows omitted from the output
collect2: error: ld returned 1 exit status
make: *** [../examples.mk:30: restful_server] Error 1
```

Thx !

syw2014, 2020-08-09:
@benrubson Have you solved it ,I meet the same errors with compile with msys2

benrubson, 2020-08-10:
Unfortunately no.
I also made another finding, mongoose 6.6 which was the last version to properly build does not compile with gcc 9.
You have to choose gcc 7, still available in Cygwin repository.
All other mongoose versions > 6.6 do not build, whatever gcc version used...

cpq, 2020-08-11:
That looks very weird.
Could you manually edit the mongoose.c source and comment out the following line:
```
int mg_vcmp(const struct mg_str *str1, const char *str2) WEAK;
```

Then recompile.
Maybe Cygwin does not like the WEAK linkage declaration, and somehow ignores the symbol in a resulting object file.

benrubson, 2020-08-11:
Thank you @cpq for your help.
Your proposal does the trick, at least with gcc 7.4.0.
With gcc 9.2.0 / 9.3.0, compilation fails, gcc returns 1 as return code, and strangely enough I don't get any message :|

cpq, 2020-08-11:
All right, thank you.
We plan to release 7.0 sometime soon - which gets rid of those WEAK things.

benrubson, 2020-08-11:
Good news, thank you :+1:
Any advice perhaps regarding gcc 9.x issues ?
Thank you again.

cpq, 2020-08-11:
No idea about the gcc 9 . If you could make it spit some error message / increase verbosity, that'll help.

benrubson, 2020-08-11:
I tried several things again, but can't manage to get an error output from gcc 9.x...
If anyone wants to try, feel free ! Thx 👍

cpq, 2020-12-17:
Closing this.

Feel free to reopen - in which case, state the OS and compiler version.

closed 2020-12-17 by cpq

Kungergely, 2021-03-02:
> Maybe Cygwin does not like the WEAK linkage declaration, and somehow ignores the symbol in a resulting object file.

Yes, this is the actual reason behind this. The linkers on Windows (both Cygwin and MSYS2 and probably MinGW too) don't like the weak symbol declaration at all and the linking of the resulting object file will fail. My solution was to open mongoose.h, find the following compiler directive:
```
#if !defined(WEAK)
#if (defined(__GNUC__) || defined(__clang__) || \
     defined(__TI_COMPILER_VERSION__)) &&       \
    !defined(_WIN32)
#define WEAK __attribute__((weak))
#else
#define WEAK
#endif
#endif
```
and replace the `!defined(_WIN32)` with `(!defined(_WIN32)&&!defined(__CYGWIN__))`. After this it compiled and linked successfully.

cpq, 2021-03-04:
Pushed https://github.com/cesanta/mongoose/commit/e710df4c80816f8406e6d943211e9adf0f400df1 - please confirm it works as expected

rajhlinux, 2022-04-18:
I'm getting the same error for compiling websocket server ".c" example using Cygwin or minGW

```
gcc -v

Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-11.2.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/include --libexecdir=/mingw64/lib --enable-bootstrap --enable-checking=release --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts --enable-libstdcxx-time --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-lto --enable-libgomp --disable-multilib --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev10, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --with-boot-ldflags='-pipe -Wl,--disable-dynamicbase -static-libstdc++ -static-libgcc' LDFLAGS_FOR_TARGET=-pipe --enable-linker-plugin-flags='LDFLAGS=-static-libstdc++\ -static-libgcc\ -pipe\ -Wl,--stack,12582912'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (Rev10, Built by MSYS2 project)
```

cpq, 2022-04-19:
@rajhlinux you forgot to add mongoose.c to the compilation command