cesanta/mongoose #913: use bind tcp listen port while event jump into  udp event（nc->flags & MG_F_UDP ）== true；

lzhong28, 2018-04-12:


lzhong28, 2018-04-12:
them http server is hanging ，the stack as：
#0  0x0000003ebd60decc in recvfrom () from /lib64/libpthread.so.0
#1  0x000000000042f534 in mg_recvfrom (nc=0x11fc8ba0, sa=0x2b3472e2c970, sa_len=0x2b3472e2c96c, buf=0x2b3472e2c980) at ../../../src/mongoose/mongoose.c:3393
#2  0x000000000042f5c3 in mg_handle_udp_read (nc=0x11fc8ba0) at ../../../src/mongoose/mongoose.c:3405
#3  0x000000000042f7c8 in mg_mgr_handle_conn (nc=0x11fc8ba0, fd_flags=1, now=1523346672.9918389) at ../../../src/mongoose/mongoose.c:3487
#4  0x00000000004300f2 in mg_socket_if_poll (iface=0x11e3e770, timeout_ms=500) at ../../../src/mongoose/mongoose.c:3690
#5  0x000000000042cd0d in mg_mgr_poll (m=0x11e3edc0, timeout_ms=500) at ../../../src/mongoose/mongoose.c:2232

cpq, 2020-12-17:
Closing this.
Feel free to reopen - but use the latest code please.

closed 2020-12-17 by cpq