cesanta/mongoose #915: Mongoose-esp32-mbedTLS

dawidbruz, 2018-04-18:
Hi, im trying to run mongoose web server on esp32, websocket_chat example, in http, it works just fine, but when i add an SSL(mbedTLS) to it(im using self signed certificate), it becomes very unstable, sometmes i can load the page, but i cant communicate via chat, and usually browser gives me ERR_CONNECTION_CLOSED or TIME_OUT, and some errors from esp debug: 

mg_ssl_if_mbed_err   0x3fff3c34 mbedTLS error: -0x4300

-0x0010

-0x004c

-0x004e

i compile it with flags: -DMG_ENABLE_SSL=1 -DMG_SSL_IF=MG_SSL_IF_MBEDTLS -DMG_ENABLE_HTTP_WEBSOCKET -DMG_ENABLE_FILESYSTEM -DESP_PLATFORM -DMG_ENABLE_HTTP_SSI=1 -DMG_NET_IF

i need websocket broadcast in ssl, but it does not want to work, any ideas?

rosimildo, 2018-05-18:
I had the same set of issues. It nearly works. 
In my case, I wrote the code in a way that I can compile on Linux and it works running as a process in Linux.
The same code on ESP32, give the errors you mentioned. In my tests, it never passes the handshake phase.

I suspect it could be a RAM issue, but I switched to a WROVER kit board, the same issues are there.

cpq, 2020-12-17:
Closing this.
There is a working ESP32 example: https://github.com/cesanta/mongoose/tree/master/examples/esp32

closed 2020-12-17 by cpq