cesanta/mongoose #923: Websockets client on ESP-IDF

janoist1, 2018-05-24:
I'm trying to compile and run the [websocket_chat_client example ](https://github.com/cesanta/mongoose/blob/master/examples/websocket_chat_client/websocket_chat_client.c)on ESP-IDF platform but the value of status on MG_EV_CONNECT event is always -1, which is a Connection error. What I've done so far is I took the [esp32-idf example](https://github.com/cesanta/mongoose/tree/master/examples/ESP32_IDF), merged it it the websocket client one, set the chat_server_url to a host where a WS server is listening.

I'm new in this area, how to start investigating? How to debug?

bakhshizadeh, 2018-12-17:
may someone give me websocket client example for esp-idf?

cpq, 2020-12-17:
Closing this.
Please use the latest esp32 example: https://github.com/cesanta/mongoose/tree/master/examples/esp32
You could integrate a websocket client example there : https://github.com/cesanta/mongoose/tree/master/examples/websocket-client

closed 2020-12-17 by cpq