cesanta/mongoose #932: Double thread and segfault

pinkra, 2018-06-13:
Hi,
I have a program running two threads with Mongoose.
One thread is to handle http connections, while the other one is for https.
On each thread I initialize Mongoose and so on (like an http/https server example).
Sometimes I got segfault during the threads initialization:

`#0  __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
#1  0x76c8a648 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#2  0x76c8b34a in __GI_abort () at abort.c:89
#3  0x76cb15a6 in __libc_message (do_abort=do_abort@entry=2, fmt=0x76d31d74 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
#4  0x76cb52ae in malloc_printerr (action=<optimized out>, str=0x76d31e44 "double free or corruption (!prev)", ptr=<optimized out>, ar_ptr=<optimized out>) at malloc.c:5006
#5  0x76cb5a1c in _int_free (av=av@entry=0x75b00010, p=<optimized out>, p@entry=0x75b004d0, have_lock=have_lock@entry=1) at malloc.c:3867
#6  0x76cb79bc in _int_realloc (av=av@entry=0x75b00010, oldp=oldp@entry=0x75b004d0, oldsize=oldsize@entry=72, nb=nb@entry=136) at malloc.c:4358
#7  0x76cb896a in __GI___libc_realloc (oldmem=0x75b004d8, bytes=128) at malloc.c:3045
#8  0x76d8c59e in CRYPTO_realloc () from /lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
#9  0x76ddb594 in lh_insert () from /lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
#10 0x76d8e2e8 in OBJ_NAME_add () from /lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
#11 0x76ea5a3a in SSL_library_init () from /lib/arm-linux-gnueabihf/libssl.so.1.0.0
#12 0x0001908e in mg_ssl_if_init () at mongoose/mongoose.c:4465
#13 mg_mgr_init_opt (m=m@entry=0x76451d30, user_data=user_data@entry=0x0, opts=...) at mongoose/mongoose.c:2353
#14 0x000190f0 in mg_mgr_init (m=m@entry=0x76451d30, user_data=user_data@entry=0x0) at mongoose/mongoose.c:2327
#15 0x00022782 in http_redirect_thread_process (thread_arg=0xff82f8 <intercom_data>) at src/wai.c:258
#16 0x76ec95b4 in start_thread (arg=0x0) at pthread_create.c:335
#17 0x76cf8c4c in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:89 from /lib/arm-linux-gnueabihf/libc.so.6`

What is wrong in my code?

psykokwak-com, 2020-11-18:
Add a mutex over all your mg_mgr_init() calls : 

> mutex.take();
> mg_mgr_init(....);
> mutex.give();

cpq, 2020-12-17:
Please use the socket pair pattern:

https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded

closed 2020-12-17 by cpq