cesanta/mongoose #934: Segmentation fault when try to close multipart upload at MG_EV_HTTP_PART_BEGIN

fangqiao, 2018-06-18:
Hi,
I am trying mongoose with your big file upload example.
I have a use case that the server should reject upload when user send multipart data with none file part.
It caused segmentation fault. Is this a bug? What is the correct way for my purpose?

I use following code:
```
  struct file_writer_data *data = (struct file_writer_data *) nc->user_data;
  struct mg_http_multipart_part *mp = (struct mg_http_multipart_part *) p;

  switch (ev) {
    case MG_EV_HTTP_PART_BEGIN: {

      if( mp->file_name == NULL || strlen(mp->file_name) == 0 ) {

     // check if there is none file part.

        printf(" reject as non-file field not allowed...");
        mg_printf(nc, "%s",
                  "HTTP/1.1 550 Only file upload allowed\r\n"
                  "Content-Length: 0\r\n\r\n");
        nc->flags |= MG_F_CLOSE_IMMEDIATELY; //MG_F_SEND_AND_CLOSE;
        free(data);
        nc->user_data = NULL; 
        return;
      }

```

fangqiao, 2018-06-18:
Further tested and found out that anytime if attempt to send response before MG_EV_HTTP_PART_END, segment fault would occur.

cpq, 2020-12-17:
Closing this, as multipart handling is not in the latest codebase anymore.

closed 2020-12-17 by cpq