cesanta/mongoose #949: Resurrect #753: support for PUT, DELETE, etc methods

Bklyn, 2018-07-19:
I'd like to revisit #753. 

What you've suggested as a solution for #753 is a pretty large amount of code for something that seems like it should be very simple.  In my code I can either:

1. Expose the private function ``mg_connect_http_base`` and reproduce the rest of ``mg_connect_http_opt`` verbatim, which is about 25-30 LOC
2. Copy the logic of ``mg_connect_http_base`` and then call ``mg_connect_opt`` and then reproduce the rest of ``mg_connect_http_opt``, so probably 50-60 LOC.

As a user of this wonderful library I'd much rather have a variant of ``mg_connect_http`` or ``mg_connect_http_opt`` that lets me pass the method explicitly.

demsking, 2019-12-10:
Any update for this issue?

LeafHCode, 2020-12-01:
> I'd like to revisit #753.
>
> What you've suggested as a solution for #753 is a pretty large amount of code for something that seems like it should be very simple. In my code I can either:
>
> 1. Expose [...]

Any update for this issue? Any pr or example?

cpq, 2020-12-17:
Closing this - as the current API is pretty must what you're asked for.

closed 2020-12-17 by cpq