cesanta/mongoose #961: mg_broadcast not working

snahmad, 2018-08-17:
```
namespace
{
	static void ev_handler(struct mg_connection *nc, int ev, void *ev_data) 
	{

		switch (ev) {
		case MG_EV_POLL:
			break;
		case MG_EV_SEND:
			break;
		case MG_EV_ACCEPT:
			break;
		case MG_EV_CONNECT:
			break;
		case MG_EV_RECV:
			break;
		case MG_EV_CLOSE:
			break;
		default:
			break;
		}
	}

}

void sendUDPMessages()
{
	ACDTRACEF(ES_NoError, "Enter sendUDPMessages \n");
	
	struct mg_mgr mgr;
	mg_mgr_init(&mgr, NULL);

	struct mg_connect_opts opts;

	memset(&opts, 0, sizeof(opts));
	opts.flags = MG_F_ENABLE_BROADCAST;
	opts.iface = nullptr;

	
	const char *port1 = "udp://1457", *port2 = "udp://127.0.0.1:1457";
	const char *port3 = "udp://255.255.255.255:1457";

	struct mg_connection * nc_client = mg_connect_opt(&mgr, port3, ev_handler, opts);	
	
	bool stop = false;
	std::thread thread = std::thread([&]() {
		while (!stop) {
			mg_mgr_poll(&mgr, 1);
		}
	});
	
	for (int i = 1; i <= 10; i++)
	{
		std::string  send_msg = "hello";
		mg_broadcast(&mgr, ev_handler, (void*)send_msg.c_str(), send_msg.size());
	}
	
	std::this_thread::sleep_for(std::chrono::seconds(3));
	
	stop = true;
	if (thread.joinable()) {
		thread.join();
	}

	mg_mgr_free(&mgr);
	
}

```
Here is my example code.

mg_broadcast not working/ doing nothing on windows platform.
On ESP32 platform it does not build. linker error with mg_broadcast

snahmad, 2018-08-19:
kindly some one reply please.

CfirTsabari, 2018-11-08:
hi,
I work on windows and i broadcast work just fine for me.
I am not sure but It's look like you don't realize what broadcast should do.
It's doesn't suppose to send any messages.
It's more like an infra structure helper function that should help you use mongoose in multi threaded application.
in this example nothing should happen.
when you call mg_broadcast this should raise an event in the mg_thread(MG_EV_POLL)

snahmad, 2018-11-20:
ok. to send udp message to 255.255.255.255 . we still need to use mg_send.
I works for me. sorry I got confuse with mg_broadcast.

cpq, 2020-12-17:
Please use https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded example.

closed 2020-12-17 by cpq