cesanta/mongoose #964: mg_mgr_poll second paramater

snahmad, 2018-08-24:
mg_mgr_poll(&mgr, ?);

My platform is Esp32

What is purpose of second paramater.
I notice if have 1 tcp connection than mg_mgr_poll(&mgr, 1); is good enough.

but I have 3 tcp connections it slows down receiving and sending tcp messages.
but i have to do mg_mgr_poll(&mgr, 2); to give more time for poll.

Is my understanding is correct?

If I  more work to do I need to give more time ?

snahmad, 2018-09-11:
Can anyone answer my question

PuKathy, 2018-11-07:
`int mg_mgr_poll(struct mg_mgr *m, int timeout_ms) {
  tv.tv_sec = timeout_ms / 1000;
  tv.tv_usec = (timeout_ms % 1000) * 1000;

  num_ev = select((int) max_fd + 1, &read_set, &write_set, &err_set, &tv);
}

while (1) {
 mg_mgr_poll(&mgr, 200);
}
`

mg_mgr_poll will use 200ms to detect events on connections。If a event is detected, mg_mgr_poll will return immediately before 200ms。

snahmad, 2019-07-30:
Can some one explain bit further. How you achieve poll using sleep or yield. Is this uses any operating system call.

cpq, 2020-12-17:
The second parameter is a number of milliseconds to block, if there is no activity on all connections.

closed 2020-12-17 by cpq