cesanta/mongoose #970: mg_send_websocket_frame issue

xyuan240, 2018-09-11:
Hi,

I am using the following piece of code to broadcast messages to client. Occasionally (not able to replicate), assertion "assert(a->len <= a->size);" in function mbuf_insert is triggered.  

%%%
   for (c = mg_next(nc->mgr, NULL); c != NULL; c = mg_next(nc->mgr, c))
   {
      if(c != nc )
         continue; 
      mg_send_websocket_frame(c, WEBSOCKET_OP_TEXT, buf, strlen(buf));
   } 
%%%

After debugging the issue, I found it was caused by mg_connection pointer c , in which its  send_mbuf has length greater than size. 

I can see the problem happens more often (but not every time) when I repeat the process of initializing a session in browser, close it and open it again a few times.

I wonder in what scenarios could cause a send_mbuf having length greater than its size? Could you provide some ideas to debug this issue further?Thanks!

cpq, 2020-12-23:
Does that manifest itself on the latest codebase?

cpq, 2020-12-27:
Closing this - feel free to reopen if that continues to happen on the latest codebase.

closed 2020-12-27 by cpq