cesanta/mongoose #971: chunked encoding client issue

schmurfy, 2018-09-14:
I use mongoose for an HTTP client, it works fine for standard requests but I need to handle streamed content with chunked encoding and I cannot make it work, I will try to make a small reproduction case but maybe might have an idea what is happenning.

The server return this:

``` http
HTTP/1.1 200 OK
Api-Version: 1.37
Content-Type: application/json
Docker-Experimental: false
Ostype: linux
Server: Docker/18.03.0-ce (linux)
Date: Fri, 14 Sep 2018 09:14:21 GMT
Transfer-Encoding: chunked

136
<DATA>

120
<DATA>
```

The connection is never closed so what I did is:
- create a connection
- use mg_set_protocol_http_websocket to set the http event handler
- define my handler for MG_EV_HTTP_CHUNK which try parse the message (json) and set the MG_F_DELETE_CHUNK flag if successfull

it mostly works but I have two issues:
- sometimes my handler receive an `http_message` which contains the bytes count (120, 136) as string, I checked the http messages received and the byte count sent by the webserver correctly matches what is sent so I don't understand why mongoose get confused. I tried to debug it myself but I failed.
- sometimes an assertion fails: `7979 >= 44506
Assertion failed: (blen >= body_len), function mg_handle_chunked, file [...]/mongoose/mongoose.c, line 6188`, I also looked at it and try to understand but I am not sure what exactly this assertion checks and how it can fails.

schmurfy, 2018-09-14:
I made a small reproduction case and I got the same results:
https://gist.github.com/schmurfy/1ee7c001553e4d5f970ec06bcb72fb6c

The server sends 10 chunks, this client correctly prints the first 8 and then the 9th is printed with the byte size for some reason, the 19th is printed correctly without the header.

I checked the real size of the message versus what is written in the chunk header and they match... Thanks for any idea or help, I am lost on this one.

schmurfy, 2018-09-18:
I switched the http parser with another one and it now works much better, I am not sure what is the issue but there is something wrong with mongoose with the way chunked encoding is handled, the server I was connecting to is the dockerd api socket.

Any input is welcome on this, it is a shame because I think mongoose is the one of the best if not the best all in one network layer I have used and I would much prefer use it alone for this task.

schmurfy, 2018-09-25:
Here is a minimal reproduction case: https://gist.github.com/schmurfy/5115723157f1b7390b51e62e2fe83da8
I included both the client and server, when I execute them I get that:

```
Assertion failed: (blen >= body_len), function mg_handle_chunked, file ../mongoose/mongoose.c, line 6193.
```

I hope this can help someone help me :)

schmurfy, 2018-09-26:
While waiting for a answer solution adding the following after a `MG_EV_HTTP_REPLY` is received does the job, it looks like the buffer is not cleared after a chunk has been fully reassembled.

```c
        if( nc->proto_data != NULL ){
          nc->proto_data_destructor(nc->proto_data);
          nc->proto_data = NULL;
        }
```

venkykrishvvk, 2018-09-26:
Can you post the modified source?
I am also facing similar issue.
I am writing a http client to collect the streaming (video) data from a server (writen by another party), using "chunked" transfer.

schmurfy, 2018-09-26:
I did not modify the source, I did it in my client code, here is the relevant part of my handler:

```c
static void ev_handler(struct mg_connection *nc, int ev, void *ev_data) {
  struct http_message *hm = (struct http_message *) ev_data;

  switch (ev) {
    case MG_EV_HTTP_REPLY:
      // do whatever you want with hm->body
      
      // and make sure previous chunks are discarded before parsing new data
      if( nc->proto_data != NULL ){
        nc->proto_data_destructor(nc->proto_data);
        nc->proto_data = NULL;
      }
      break;
  }
}
```

I don't expect this to be bullet proof and this is probably not the correct fix but that is the best I could come up with and it fixed my issues.

rojer, 2018-09-26:
@schmurfy thank you for your detailed analysis, i am going to take a look at this myself now

rojer, 2018-09-26:
the real issue here is with keepalive - connection state is not properly cleared between responses.
i sent out a fix.

venkykrishvvk, 2018-09-27:
Have you already committed? can I see the fix?
My server returns:
HTTP/1.1 200 OK
Connection: Keep-Alive
Transfer-Encoding: chunked
Accept-Ranges: none
Content-Type: video/mp2t
Date: Sat, 01 Jan 2000 04:05:53 GMT
1FF4/r/n
DATA-1/r/n
1FF4/r/n
DATA-2/r/n
..
..
so on so forth.

But I get the MG_EV_RECV for much less data len 0x5B4 for example.
I am setting mg_set_protocol_http_websocket(nc);
I handle MG_EV_HTTP_CHUNK in my event handler, but the data is not proper.
I do set the delete chunk flag in the event handler after consuming the data.

Its unfortunate, we dont have an example code for server and client using http chunked transfers.

FYI, my server is sending continues Live streaming video data (raw Transport stream), using chunked transfers.

I validated the server side using a curl client.

rojer, 2018-09-27:
@venkykrishvvk not yet. the fix is out for review, you will see it in this commit once it lands.

schmurfy, 2018-09-27:
Thanks for looking into this, I think mongoose is a really great library :)

closed 2018-09-27 by rojer