cesanta/mongoose #974: Detected as Malware/Riskware

rabimba, 2018-09-20:
Hi,

The prebuilt windows binary is being detected as virus/malware/riskware by multiple engines including Malwarebytes Anti-malware.
In virustotal 21/66 is detceting it as bad actor/trojan.

It would be great to get these false positives removed

https://www.virustotal.com/#/file/d07955c619bb06411b3e8587883b703c4d3dd25571301d3ce4d28686a2081248/detection

rojer, 2018-09-20:
we tried once, but it just keeps creeping back up. windows anti-malware situation is a mess and we just gave up.

mica, 2019-03-17:
Windows Defender is now quarantining this immediately upon downloading.

cpq, 2019-03-17:
Thanks @mica.
We don't have control over that unfortunately.

israsource, 2019-06-23:
That's probably because code reuse, it seems your program is being used in different ways.

This file:
C:\Users\WMJI\Desktop\dyaJSXW9MS.exe

israsource, 2019-06-23:
https://analyze.intezer.com/#/analyses/38050fbe-e67e-48f6-b045-827ee6f41dec/

mtjmohr25, 2020-02-17:
Well,

mongoose.exe in its free trial form version 6.9 downloaded as of 2020-02-17 display 11/49 hits on VirusTotal. For a product that well documented and prepared with love, that is INACCEPTABLE. Furthermore inacceptable are the sloppy answers given as to why this is so.

A blockchain for compilation either leads to a clean exe, eve if compressed by UPX or whatever other compressor or does not. Either way, it must be known and explainable in case of - as here - justified questions.

And on top, the users' questions just show their positive security concern and do not reflect, as quoted, a "paranoid" behavior. That does not meet the issue at all.

Personally, I do not doubt the excellence of Mongoose at all as one can see from many other projects its perfect integration. But the question, unanswered so far, remains why the compiled exe shows this marked result on VirusTotal.

This is counterproductive for the product and the company / distributor.

oparviai, 2020-07-29:
> Personally, I do not doubt the excellence of Mongoose at all as one can see from many 
> other projects its perfect integration. But the question, unanswered so far, remains why 
> the compiled exe shows this marked result on VirusTotal.

I suppose that the issue is that while mongoose .exe file is clean itself, it's being (mis)used as part of 3rd party malware toolkits and due to being part of those 3rd party malware toolkits it quickly ends up getting flagged as malware by security analysis companies. Tough situation for the original tool developers indeed.

cpq, 2020-07-29:
@oparviai makes sense.
Well, we plan to open source not only the library source code (which is already open), but the binary's source, too. It won't stop the false positives, but at least those concerned could inspect and build the binary themselves.

cpq, 2020-12-17:
The binary is open source now: https://github.com/cesanta/mongoose/tree/master/examples/desktop-server

Closing this.

closed 2020-12-17 by cpq

mmortal03, 2021-10-16:
> The binary is open source now: https://github.com/cesanta/mongoose/tree/master/examples/desktop-server
> 
> Closing this.

Why does the fact that it became open source affect this issue? The owner of the mongoose.ws website should download the file in Microsoft Edge and when it says "mongoose.exe was blocked because it could harm your device, then they should "report this file as safe", and fill out the form after selecting "I am the owner or representative of this website and I want to report an incorrect warning about it".

open-se, 2021-10-20:
A fresh Windows 10 VM with only mongoose.exe installed leads to trojans being placed in the windows temp folder within the first 24 hours. Seems there may be a security vulnerability here leading to backdoor access even if not intended.

catrius, 2021-11-07:
[image]
I'm having the exact issue with version 7.3.3. FYI, I had version 7.3.2 on my machine, it reminded me that there was a new version, after I clicked on "Update" Windows defender immediately report mongoose.exe as a Trojan.

cpq, 2021-11-09:
Mongoose web server binary is not distributed by us anymore.

All previous versions of binary did not have any malicious code - all alerts are false positives.
If in doubt, however - just don't use the binary.

callahanp, 2025-02-15:
Seems they are at it again.  Without the .exe, what exactly are they objecting to? 

Malwarebytes Browser Guard Logo
Website blocked due to trojan
Website blocked: https://mongoose.ws
v3.0.11 | Trojan: 2.0.202502110905

Malwarebytes Browser Guard blocked this page because it may contain malicious activity.

Warning icon
We strongly recommend you do not continue. You may be putting your safety at risk by visiting this site. For more information, visit [ Malwarebytes Support.](https://links.malwarebytes.com/support/browserguard/?guard=1&x-source=support)

callahanp, 2025-02-15:
Seems they are at it again.  What is triggering this if the executable is no longer present?

Unable to reopen issue (permission needed to re-open it) so I'll create a new one.

Malwarebytes Browser Guard Logo
Website blocked due to trojan
Website blocked: https://mongoose.ws
v3.0.11 | Trojan: 2.0.202502110905

Malwarebytes Browser Guard blocked this page because it may contain malicious activity.

Warning icon
We strongly recommend you do not continue. You may be putting your safety at risk by visiting this site. For more information, visit [ Malwarebytes Support.](https://links.malwarebytes.com/support/browserguard/?guard=1&x-source=support)

scaprile, 2025-02-15:
Please don't necropost
Please don't open a new issue, this is not something we'll address.

mica, 2025-02-16:
Why are you saying it's "not distributed by us anymore"?.. It's right here: 

https://github.com/cesanta/binary/releases/download/exe/mongoose.exe

via https://mongoose.ws/binary/