cesanta/mongoose #979: esp32 webserver response large file failed

philar, 2018-10-09:
plaform:ESP32 ,do not use file system.
Try to response large file(175KB) use function "mg_send " in the event MG_EV_HTTP_REQUEST ,the file will been truncated, beacause the  MBUF_REALLOC  will failed when the file is larger than 113777 bytes. 

since the event_handle is single-threaded ,it can not real send the data when the event_handle not return ,so not matter how large the sendbuf it is ,it will call this problem.
I try to use multithread, but  mg_start_thread  do not support the ESP32

can anyone  have the same problem?

cpq, 2018-10-09:
Use chunked encoding, and stream the data out.

closed 2018-10-09 by cpq

philar, 2018-10-09:
It can not work ,I try this way before . Not matter "mg_send_http_chunk" or "mg_send" , it will call the "mbuf_insert" finally. So the sendbuf will been full . beacause there is not thread to send out the data(clear the sendbuf). Only when the event_handle return , the mg_mgr_poll will clear the sendbuf.