cesanta/mongoose #980: HTTPS server context reuse problem with client certificates

EugeneRyabtsev, 2018-10-09:
Trying to use Mongoose 6.11 as HTTPS server built in myapp on Astra Linux Orel 2.12, setting it up like so:

```
    mg_mgr_init(&mg_manager, this);
    mg_bind_opts bind_opts = { 0 };
    bind_opts.ssl_cert = "certs/Eugene_HTTPS_test_server1.crt";
    bind_opts.ssl_key = "certs/Eugene_HTTPS_test_server1.pem";
    bind_opts.ssl_ca_cert = "certs/Eugene_test_CA1.crt"; // This causes problems
    http_mg_conn = mg_bind_opt(&mg_manager, "0.0.0.0:443", http_request_handler, bind_opts);
```

Everything is working fine without ssl_ca_cert. With ssl_ca_cert only the first connection works (this might include multiple files, because keep-alive), then maybe after I close/reopen the browser or wait for a long time it works once more etc. I managed to sketch a "fix" like so:

1. Declare global variable `struct mg_ssl_if_conn_params params;`, comment out it's local equivalent in `mg_bind_opt`.
2. In `mg_accept_conn` do the following:
```
  if (lc->flags & MG_F_SSL) {
    const char *msg;
    if (mg_ssl_if_conn_init(nc, &params, &msg) != MG_SSL_OK) mg_close_conn(nc);
    //if (mg_ssl_if_conn_accept(nc, lc) != MG_SSL_OK) mg_close_conn(nc);
  } else
```
3. In `mg_ssl_if_conn_init` do the following:
```
  if (nc->flags & MG_F_LISTENING) {
    ctx->ssl_ctx = SSL_CTX_new(SSLv23_server_method());
  } else {
    ctx->ssl_ctx = SSL_CTX_new(SSLv23_server_method());
    //ctx->ssl_ctx = SSL_CTX_new(SSLv23_client_method());
  }
```

After this ugly hack, HTTPS server works as intended, accepting the client certificate, serving pages etc. It seems there is a SSL context reuse problem (it gets copied from the listener socket in `mg_ssl_if_conn_accept` and created anew in `mg_ssl_if_conn_init`).

chicoze, 2020-02-19:
This problem is still in version 6.17. I tried and I have the exact same problem!

chicoze, 2020-02-19:
I tried compiling with MbedTLS and everything works as expected.

cpq, 2020-12-17:
Closing this.
BTW, the OpenSSL code has a session reuse snippet.
Please use the latest codebase.

closed 2020-12-17 by cpq