cesanta/mongoose #990: Make SSL min version configurable

zdila, 2018-12-05:
Please make SSL min version configurable via `build_vars`.

https://github.com/cesanta/mongoose/blob/95afa323c0e7740151f661629616daebe2915b61/src/mg_ssl_if_mbedtls.c#L113

rojer, 2018-12-11:
any particular reason? do you have ancient clients or something?

zdila, 2018-12-11:
> do you have ancient clients or something?

Yes we have, unfortuntlely.

rojer, 2018-12-11:
so i thought.
well, i am really reluctant to do it. maybe if there is more demand... but ultimately, TLS 1.2 is a reasonable baseline, older versions open pandora's box of various security issues.
i'm going to push back on this for now, since we distribute in source form, you should be able to do it in your build.