cesanta/mongoose #1006: mongoose.h does not compile with i686-w64-mingw32-g++

merovingien, 2019-02-22:
Hello,

I try to use mongoose on windows with i686 mingw-w64 compiler, gcc 6.3, and it fails to compile due to a bad check of macro in mongoose.h :

Tested with mongoose 6.13 :

i686-w64-mingw32-g++ -c mongoose.h
mongoose.h:225:13: error: redeclaration of C++ built-in type 'bool' [-fpermissive]

Failure is caused by these lines : 

```c++
#if _MSC_VER < 1700
typedef int bool;
#else
#include <stdbool.h>
#endif
```

When used with mingw-w64 g++, _WIN32 is defined but _MSC_VER is not, so typedef int bool; is executed.

This workaround works for me :

```c++
#if defined(_MSC_VER) && _MSC_VER < 1700
typedef int bool;
#else
#include <stdbool.h>
#endif
```

I could not find a workaround that does not modify mongoose.h.

Thanks for your attention.

jermp, 2020-03-22:
Yes, I confirm this. I've just got the very same error.
Please check this.

binupr, 2020-10-07:
I am running into the same error for bool in mongoose latest version 6.18. I have the below code in mongoose.h

#if defined(_MSC_VER) && _MSC_VER < 1700
typedef int bool;
#else
#include <stdbool.h>
#endif

Any idea why I should run into the same issue on the latest version? I don't see problem with below code on mongoose version 6.13 though.
#if _MSC_VER < 1700
typedef int bool;
#else
#include <stdbool.h>
#endif

cpq, 2020-12-24:
What is your OS and compiler?

cpq, 2020-12-25:
Thanks everyone,
mingw gcc and g++ tests were added to the continuous integration.
Closing this.

closed 2020-12-25 by cpq