cesanta/mongoose #1013: Mqtt Broker Doesn't Pong Back

vinicius-cerebrasc, 2019-03-22:
I'm using mongoose as a component on esp-idf, so I can use ESP32 as a Broker.

`if ((nc2 = mg_bind(&mgr, s_listening_address, mqtt_broker_ev_handler)) == NULL) {
        ESP_LOGE(TAG, "mg_bind(%s) failed", s_listening_address);
        return;
    }
    mg_mqtt_broker_init(&brk, NULL);
    nc2->priv_2 = &brk;
    mg_set_protocol_mqtt(nc2);
    ESP_LOGI(TAG, "MQTT broker started on %s", s_listening_address);`

I noticed that the broker is not responding to keepalive ping requests, eventually losing connection with the client (I'm using MQTT.fx).

> I (215101) MONGOOSE: BROKER HANDLER GOT EVENT MG_EV_RECV
> 
> I (215101) MONGOOSE: BROKER HANDLER GOT EVENT MG_EV_MQTT_PINGREQ
> 
> I (275181) MONGOOSE: BROKER HANDLER GOT EVENT MG_EV_CLOSE

Also, looking inside mongoose.c, the pong function isn't being called anywhere else, apart from it's declaration.
`void mg_mqtt_pong(struct mg_connection *nc) {
  mg_send_mqtt_header(nc, MG_MQTT_CMD_PINGRESP, 0, 0);
}`
Any sugestions?

PuKathy, 2019-05-23:
In "mqtt_broker_ev_handler", handle  "MG_EV_MQTT_PINGREQ" with "mg_mqtt_pong()".
Also, Qos events are not been handled.

nopnop2002, 2019-07-10:
You have to send MQTT_PINGRESP using "mg_mqtt_pong()".
Please refer this:
https://github.com/nopnop2002/esp-idf-mqtt-broker

closed 2019-07-25 by vinicius-cerebrasc