cesanta/mongoose #1015: Just curious if it is possible to run mongoose on mips

linuxenko, 2019-03-24:
Hi :unicorn: 

It is one of the craziest `mips`es I've ever seen, it has no `pthreads` and super duper events support. Just plain old selects and forks )) So I decided to try run `mongoose` on it )) It seems like an appropriate option for such hardware )) 

The problems started at the beginning of this "journey", with following output:

```
➜  mips-gcc ../../mongoose.c -I../.. -shared
../../mongoose.c:1558: warning: weak declaration of 'mbuf_resize' after first use results in unspecified behavior
../../mongoose.c:1693: warning: weak declaration of 'mg_mk_str' after first use results in unspecified behavior
../../mongoose.c:1821: warning: weak declaration of 'mg_str_starts_with' after first use results in unspecified behavior
../../mongoose.c:2212: warning: weak declaration of 'mg_avprintf' after first use results in unspecified behavior
../../mongoose.c:2270: warning: weak declaration of 'mg_next_comma_list_entry_n' after first use results in unspecified behavior
```

The gcc is the uClibc crosscompiler toolchain (uclibc v0.9.30.1) from their website, it compiles my procets I've been working/experimenting with. 

```
uClibc mips-rawgcc (GCC) 4.1.2
```

And, it dies yelling "interuption error" or something like this when I ran it on the device. Just curious if it possible to run it there, maybe I didn't notice a build option or something.. ? Thanks if you have a time to help me )) 
:steam_locomotive: :steam_locomotive: :steam_locomotive: :steam_locomotive: :steam_locomotive: :steam_locomotive: :steam_locomotive:

cpq, 2019-03-24:
Should work. I guess that platform supports posix and BSD sockets.
First try to disable everything possible, see all possible definitions by running 

```
perl -nle 'print $1 if /^#if (MG_\S+)/' mongoose.c | sort | uniq
```

linuxenko, 2019-03-24:
Шас спою )) I love the cartoon )

Seems like options do not help with the problem.
Sorry for my `sed`
```
perl -nle 'print $1 if /^#if (MG_\S+)/' mongoose.c | sort | uniq | sed 's/.*/-D\0/g' | for i in `xargs`; do echo "Building with $i"; mips-gcc mongoose.c -shared $i; done &> out
```

[The "fuzzing" output](https://github.com/b37t1td/trqbp/blob/master/out)

It feels like the "warnings" is an messengers of unsuccessful builds, I didn't run every of them on the device, but it really looks like nothing changed.

linuxenko, 2019-03-29:
Finally found some time to report it properly 

I think, it does not require any explanation:
```
mips-gcc echo_server.c ../../mongoose.c -o echo_server -g -W -Wall -Werror -I../.. -Wno-unused-function
cc1: warnings being treated as errors
../../mongoose.c:1558: warning: weak declaration of 'mbuf_resize' after first use results in unspecified behavior
../../mongoose.c:1693: warning: weak declaration of 'mg_mk_str' after first use results in unspecified behavior 
../../mongoose.c:1821: warning: weak declaration of 'mg_str_starts_with' after first use results in unspecified behavior
../../mongoose.c:2270: warning: weak declaration of 'mg_next_comma_list_entry_n' after first use results in unspecified behavior 
```

Ok then, lets try without strict flags:

```
mips-gcc echo_server.c ../../mongoose.c -o echo_server -g -W  -I../.. -Wno-unused-function
../../mongoose.c:1558: warning: weak declaration of 'mbuf_resize' after first use results in unspecified behavior
../../mongoose.c:1693: warning: weak declaration of 'mg_mk_str' after first use results in unspecified behavior 
../../mongoose.c:1821: warning: weak declaration of 'mg_str_starts_with' after first use results in unspecified behavior
../../mongoose.c:2212: warning: weak declaration of 'mg_avprintf' after first use results in unspecified behavior
../../mongoose.c:2270: warning: weak declaration of 'mg_next_comma_list_entry_n' after first use results in unspecified behavior
```

Yahoo! we got the binary `echo_server`, lets run it inside of the `mips` environment:

```
./echo_server: can't resolve symbol '__sysv_signal'
```

Aha.. there are no such calls support .. How can I fix it ? (rebuild uClibc is not an option )
Thanks )
:owl: :owl: :owl:

cpq, 2019-03-30:
Define your own dummy `signal` function, i guess. The linker should pick it up instead of the libc's `signal`.

linuxenko, 2019-03-30:
Unfortunately it has crashed when I ran it on the device. But ! it runs just fine inside of the environment built upon qemu. Which debug options could you recommend ? gdb and jtags does not work for me, it is too expensive and requires soldering skills :) Thanks 

:turtle: :turtle::turtle:

cpq, 2019-03-30:
The approach i usually take is to stick printfs all around, and narrow it down to a place of a crash.
Is your system capable of showing the printf logs ?

cpq, 2019-06-13:
Closing for now, please reopen if required.

closed 2019-06-13 by cpq