cesanta/mongoose #1016: Connection error for some users caused by invalid DNS IP fetched from registry

MexUK, 2019-04-09:
The mg_get_ip_address_of_nameserver function can give an output IP that has trailing data on the end of it.
This is becasue the specification page for RegQueryValueExW says:

"If the data has the REG_SZ, REG_MULTI_SZ or REG_EXPAND_SZ type, the string may not have been stored with the proper terminating null characters.
Therefore, even if the function returns ERROR_SUCCESS, the application should ensure that the string is properly terminated before using it; otherwise, it may overwrite a buffer."

This can probably be solved by a wmemset, sanitizing the IP from RegQueryValueExW, or both.

yang-zzhong, 2019-07-02:
some system use space replace common to split dnses
```
//wchar_t *comma = wcschr(value, ',');
wchar_t *comma = wcschr(value, ' ');
if (comma != NULL) {
     *comma = '\0';
}
```

cpq, 2019-07-02:
Thanks @MexUK and @yang-zzhong ! It is still unclear a bit, what is the issue and a suggested solution. If you follow up with a quick example, it'd be appreciated.

yang-zzhong, 2019-07-02:
@cpq in function mg_resolve_async_opt declared
```
  char dns_server_buff[17]
```
to take a dns ip. it's enough if mg_get_ip_address_of_nameserver do the right thing. but it's not. on my workmate's PC running Windows 7 ultimate, dns ip addresses are not seperated by comma but a space. so the following code work not correct.
```
        /*
         * See https://github.com/cesanta/mongoose/issues/176
         * The value taken from the registry can be empty, a single
         * IP address, or multiple IP addresses separated by comma.
         * If it's empty, check the next interface.
         * If it's multiple IP addresses, take the first one.
         */
        wchar_t *comma = wcschr(value, ',');
        if (comma != NULL) {
          *comma = '\0';
        }
```
I don't know why.

nliviu, 2019-07-02:
https://github.com/cesanta/mongoose/blob/d5beb7ba3f3767891f3d85945d7d33c1d8596e37/src/mg_resolv.c#L55-L61
The keys `NameServer` and `DhcpNameServer` have different separators:
`NameServer` uses comma and `DhcpNameServer` uses space.
Checked on Windows 7 and Windows 10.

cpq, 2020-12-23:
Closing this. Please use the latest code.
From version 7.0 on, Mongoose does not read `/etc/resolv.conf` file (or registry on Win32).

closed 2020-12-23 by cpq