cesanta/mongoose #1018: Failing to load files with latest chrome build

bemineni, 2019-04-22:
We are seeing some file loading failures with mongoose on the latest Chrome build net:Err_empty_response
Chrome Version 73.0.3683.103 (Official Build) (64-bit)

```
static enum mg_ssl_if_result mg_ssl_if_ssl_err(struct mg_connection *nc,
                                               int res) {
  struct mg_ssl_if_ctx *ctx = (struct mg_ssl_if_ctx *) nc->ssl_if_data;
  int err = SSL_get_error(ctx->ssl, res);
  if (err == SSL_ERROR_WANT_READ) return MG_SSL_WANT_READ;
  if (err == SSL_ERROR_WANT_WRITE) return MG_SSL_WANT_WRITE;
  **DBG(("%p %p SSL error: %d %d", nc, ctx->ssl_ctx, res, err));**
  nc->err = err;
  return MG_SSL_ERROR;
}
```
000000E7E28D55F0 SSL error: -1 1
We were using mongoose 6.7, but we did a test with the latest mongoose server and we still see this issue.
Have anyone see a similar issue with the latest Chrome build?.

cpq, 2019-06-13:
Does it manifest itself without SSL?

Tim5000, 2019-11-03:
Hi has there been anymore updates on this?
I believe I'm also seeing this using Mongoose 6.12.
It seems fine with HTTP (in my case) but I get lots of handshaking issues and partially loaded pages. I debugged down far enough to see handshaking issues.
Do you think pull request #877 could be related?
Thanks

cpq, 2020-12-23:
Please use the latest code - and reopen if required.

closed 2020-12-23 by cpq