cesanta/mongoose #1021: udp receive not data on Windows platform.

snahmad, 2019-05-10:
tcp is working but udp not.

I do get MG_EV_CONNECT and MG_EV_CLOSE, but no MG_EV_RECV

I have tried all these optons

char url[256];
		//sprintf(url, "udp://0.0.0.0:%s", port_.c_str());
		//sprintf(url, "udp://255.255.255.255:%s", port_.c_str());
		sprintf(url, "udp://%s", port_.c_str());
		//sprintf(url, "%s:%s", ip_address_.c_str(), port_.c_str());
		auto nc_ = mg_connect_opt(&mg_mgr_pimpl_->mgr_, url, ev_handler, opts_connect);
              auto nc_ = mg_connect(&mg_mgr_pimpl_->mgr_, url, ev_handler);
		/*
		struct mg_bind_opts opts;
		memset(&opts, 0, sizeof(opts));
		opts.flags = MG_F_ENABLE_BROADCAST;
		opts.iface = nullptr;

		//mg_mgr_pimpl_->nc_ = mg_bind_opt(&mg_mgr_pimpl_->mgr_, url, ev_handler, opts);
		mg_mgr_pimpl_->nc_ = mg_bind(&mg_mgr_pimpl_->mgr_, url, ev_handler);

snahmad, 2019-05-10:
I can see udp data in WireShark broadcast on 255.255.255.255 address.

cpq, 2020-12-23:
There is no way for UDP not working, because Mongoose has unit tests that run on Mac, Linux, Windows.
All tests use DNS resolution which is over UDP. That proves that UDP works.
Unit test are triggered on every commit, and results are available at "Actions" Github tab.

closed 2020-12-23 by cpq