cesanta/mongoose #1023: How to parse the POST parameters in JSON format with Content-Type=application/json ?

edem, 2019-05-16:
If client side send a request as below:
curl -d '{"key1":"value1", "key2":"value2}' -H "Content-Type: application/json" http://xxx.xxx.xxx.xxx:8000

How does the mongoose read the post arguments?

nliviu, 2019-05-16:
Parse the message body using [frozen](https://github.com/cesanta/frozen) or [mjson](https://github.com/cesanta/mjson)

edem, 2019-05-20:
Excuse me, how do I get the JSON parameter from POST before I parsing it?
I can get POST parameters with "mg_get_http_var(&hm->body, ...".
I can get GET parameters with "mg_get_http_var(&hm->query_string, ..."
However, I don't know how to get parameters in "application/json" Content-Type.

nliviu, 2019-05-20:
`hm->body` is the json string in the POST query.
The only way to get the arguments is to parse `hm->body` with a json parser.

closed 2019-05-23 by edem