cesanta/mongoose #1034: HTTP requests take >250s on ESP32 after updating from 6.14 to 6.15

avanbremen, 2019-07-02:
**The issue**
After updating Mongoose from version 6.14 to 6.15 (latest), HTTP requests take a very long time (over 250 seconds) on the ESP32 (ESP-IDF v4.0-dev). Besides updating Mongoose, no changes were made.

**Expected behaviour**
HTTP requests should finish in mere seconds, as is the case when using Mongoose version 6.14.

**Actual behaviour**
HTTP requests finishing in >250 seconds.

**Compile definitions**
-DMG_ENABLE_BROADCAST=1
-DMG_ENABLE_FILESYSTEM=1
-DMG_ENABLE_HTTP_STREAMING_MULTIPART=1
-DMG_ENABLE_SSL=1
-DMG_SSL_IF=MG_SSL_IF_MBEDTLS

Please let me know if there is anything else you need or anything I can do to help close this issue.

Thank you and kind regards,
Arjan

nliviu, 2019-07-02:
Can you provide a simple example to reproduce this issue?

avanbremen, 2019-07-03:
I will try to create an example project with latest esp-idf and tools as soon as possible, similar to what I created for issue [899](https://github.com/cesanta/mongoose/issues/899). However, the changes from 6.14 to 6.15 are pretty minimal. Could it have something to do with the `LWIP_TCPIP_CORE_LOCKING` addition?

I guess it doesn't seem likely, as the define is set to 0 in lwipopts.h (esp-idf/components/lwip/port/esp32/include/lwipopts.h).

```
/**
 * LWIP_TCPIP_CORE_LOCKING: (EXPERIMENTAL!)
 * Don't use it if you're not an active lwIP project member
 */
#define LWIP_TCPIP_CORE_LOCKING         0
```

avanbremen, 2019-07-15:
Please find an example project [on GitHub here](https://github.com/avanbremen/esp-mg-issue-1034). A detailed description can be found in [mg_test_main.c](https://github.com/avanbremen/esp-mg-issue-1034/blob/master/main/mg_test_main.c).

avanbremen, 2019-07-22:
@cpq Could you please take a look at the example project on [GitHub](https://github.com/avanbremen/esp-mg-issue-1034/blob/master/main/mg_test_main.c)? It can be used to easily reproduce the issue. Hopefully the fix will be just as easy. Kind regards.

avanbremen, 2019-09-02:
@cpq @nliviu I just updated ESP-IDF to the latest version (v4.1-dev-141-ga7e8d87d3), unfortunately the issue with Mongoose 6.15 still exists and can easily be reproduced using [this](https://github.com/avanbremen/esp-mg-issue-1034/blob/master/main/mg_test_main.c) example project. It would be great if someone could look into this.

cpq, 2020-12-24:
@avanbremen see https://github.com/cesanta/mongoose/tree/master/examples/esp32
Both client and server added there. Though the fetched URL is http, not https, the fetch time is pretty adequate.

closed 2020-12-24 by cpq