cesanta/mongoose #1036: How to specify a custom 404 web page in c?

Evi5, 2019-07-08:
I want to define 404 page for simplest_web_server, and there is no idea after google. thks~

nliviu, 2019-07-09:
If the requested uri from `struct http_message` does not exist, serve your custom 404 file.

Evi5, 2019-07-09:
Thks for help, nliviu.
some code like this for some laters with the same problem.

static void ev_handler(struct mg_connection * nc, int ev, void *p) {
    switch (ev) {
    case MG_EV_HTTP_REQUEST: {
        struct http_message *hm = (struct http_message *) p;
        if (mg_vcmp(&hm->uri, "/login") == 0) {
            login_handler(nc, ev, p);
            // handle_sum_call(nc, hm);
        } else if(mg_vcmp(&hm->uri, "/main") == 0){
            mg_serve_http(nc, (struct http_message *) p, s_http_server_opts);
        } else{
            404..
       }
   }
}

closed 2019-07-09 by Evi5