cesanta/mongoose #1037: How can I send byte data in server?

z13974509906, 2019-07-10:
refer to [example](https://cesanta.com/docs/http/server-example.html),  I write a http server:
```c++
#include <iostream>
#include "mongoose.h"

static const char *s_http_port = "5000";

static void ev_handler(struct mg_connection *c, int ev, void *p) {
  std::cout << "ev code is : " << ev << std::endl;
  if (ev == MG_EV_HTTP_REQUEST) {
    struct http_message *hm = (struct http_message *) p;
    
    std::cout << "body length is : " << hm->body.len << std::endl;
    std::cout << "body is : " << hm->body.p << std::endl;
    fwrite(hm->body.p, 1, hm->body.len, stdout);
    
    mg_send_head(c, 200, 11, "Content-Type: text/plain");
    mg_send(c, hm->body.p, hm->body.len);
  }
}

int main(void) {
  struct mg_mgr mgr;
  struct mg_connection *c;

  mg_mgr_init(&mgr, NULL);
  c = mg_bind(&mgr, s_http_port, ev_handler);
  mg_set_protocol_http_websocket(c);

  for (;;) {
    mg_mgr_poll(&mgr, 1000);
  }
  mg_mgr_free(&mgr);

  return 0;
}
```
the server receive data and send it back.

when the data is char, every thing is ok, but when is byte, the client receive data is total wrong, how can I fix it?  wish for help

z13974509906, 2019-07-11:
if I use flask as a server, it can reply right byte data, so has mongoose apple some interface to send the raw data back?

z13974509906, 2019-07-11:
my send data is: 
```c++
\x32\x00\x00\x00\x04\x69\x6d\x61\x67\x65\x00\x26\x00\x00\x00\x01\x30\x00\x00\x00\x00\x00\x00\x00\xf0\x3f\x01\x31\x00\x00\x00\x00\x00\x00\x00\x00\x40\x01\x32\x00\x00\x00\x00\x00\x00\x00\x08\x40\x00\x00
```
then I found server only return '\x32', maybe it see '\x00' as terminate char?

z13974509906, 2019-07-11:
send the data one by one can solve the problem
```c++
    mg_send_head(c, 200, (int)hm->body.len, "Content-Type: text/plain");
    for (int i = 0; i < (int)hm->body.len; ++i){
        mg_printf(c, "%c", (std::uint8_t)hm->body.p[i]);
    }
```

Is there any other effective ways?

zhangxu-ai, 2019-11-05:
the same problem to your,and have no idea

cpq, 2020-12-24:
Please use:

```c
mg_http_reply(c, 200, "Content-Type: text/plain\r\n", "hi! this is a format string: %d %s", 123, "boo");
```

closed 2020-12-24 by cpq