cesanta/mongoose #1041: PIC 32 build problems with 6.12 and up

rodgerbrennan, 2019-07-17:
I'm trying to build a project with mongoose 6.15 for PIC32 and running into build issues. 6.10, and 6.11 build correctly and work, I just had some other issues specifically with parsing headers hence moving to the current version.

But with 6.12 or greater, I get the following error messages from the XC32 compiler
`undefined reference to 'mg_if_sent_cb'`
`undefined reference to 'mg_if_recv_tcp_cb'`

Problems occur in `mg_handle_send` which calls `mg_if_sent_cb` and `mg_handle_recv` which calls `mg_if_recv_tcp_cb` and then `mg_recv_common` (if it still existed).  

I dug into this a bit and these three functions (`mg_if_sent_cb`, `mg_if_recv_tcp_cb`, `mg_recv_common`) appear to have been removed with the "**Mongoose net_if and ssl_if refactoring**" commit [#3e33e57](https://github.com/cesanta/mongoose/commit/3e33e577079b4c44af15a53d5bbf2acfd723ac6a) between the 6.11 and 6.12 releases.

I'm not sure if the correct approach is to add these functions back to the code, which results in a successful build or should `mg_handle_send` and `mg_handle_recv` be updated to conform to the refactor?

cpq, 2020-12-23:
Could you elaborate on your build environment please?

cpq, 2021-07-20:
No response, closing this

closed 2021-07-20 by cpq