cesanta/mongoose #1042: cookis_auth have bug

ZhangPengchao, 2019-07-19:
The function login_handler will be called with event MG_EV_CLOSE,and *p is NULL,then it will cause segment fault.It should judge the event like this:
    if(ev != MG_EV_HTTP_REQUEST){
        return;
    }

cpq, 2019-07-19:
Could you be more specific, where exactly the issue is, please?

atchutameka, 2019-07-26:
@cpq I believe the issue is with login_handler and logout_handler. They get called with a NULL argument for **void *p** and without a NULL check, it will crash.

I believe @ZhangPengchao  is saying we need a check like this:
```c
static void login_handler(struct mg_connection *nc, int ev, void *p) {
  struct http_message *hm = (struct http_message *) p;
  if (hm!=NULL && mg_vcmp(&hm->method, "POST") != 0) {
```

```c
static void logout_handler(struct mg_connection *nc, int ev, void *p) {
  struct http_message *hm = (struct http_message *) p;
  char shead[100];
  snprintf(shead, sizeof(shead), "Set-Cookie: %s=", SESSION_COOKIE_NAME);
  mg_http_send_redirect(nc, 302, mg_mk_str("/"), mg_mk_str(shead));
  if(hm!=NULL){
      struct session *s = get_session(hm);
 
```

cpq, 2020-12-07:
Please see https://github.com/cesanta/mongoose/pull/1060

closed 2020-12-07 by cpq