cesanta/mongoose #1046: user_data lost for multipart messages

MJLHThomassen, 2019-08-23:
When receiving a MG_EV_HTTP_MULTIPART_REQUEST (and having user_data enabled obviously), the pointer I receive in the handler is valid to an object which i passed to the  mg_register_http_endpoint(nc, uri_path, handler, user_data) method. However, the next time the handler is called, (with ev being MG_EV_HTTP_PART_BEGIN), the user_data is a NULL pointer.

I suspect this has something to do with the fact that when parsing multipart messages, the `mg_call` function is called inside of the `mg_http_multipart_call_handler `method with the user_data set to the user data of the network context `c->user_data` instead of the user_data that is coupled to the endpoint using mg_register_http_endpoint method.

Is this intended behavior? In my application i require the user_data to be a valid pointer to my own structure for custom parsing of multipart requests. The workaround i have now is to assign nc->user_data = my_own_user_data; when the MG_EV_HTTP_MULTIPART_REQUEST event calls my handler to have it available in subsequent calls with events MG_EV_HTTP_PART_BEGIN, MG_EV_HTTP_PART_DATA and MG_EV_HTTP_PART_END.

cpq, 2020-12-23:
The multipart handling API is gone from the latest codebase.
For file uploads, use alternative method - see https://github.com/cesanta/mongoose/tree/master/examples/file-upload

closed 2020-12-23 by cpq

MJLHThomassen, 2020-12-23:
@cpq 

Unfortunately, that API does not work for me. I don't want to write incoming data to disk. I currently use the multipart request for a firmware update. 

When i receive MULTIPART_REQUEST_MESSAGE_TYPE_PART_BEGIN, i start an OTA update of my device. At every MULTIPART_REQUEST_MESSAGE_TYPE_PART_DATA, i directly write the incoming data to the mcu's flash memory. This does not work with a file system but with custom OTA update methods.

Then at MULTIPART_REQUEST_MESSAGE_TYPE_PART_END i end the firmware update, set the new partition for next boot and reboot my mcu. 

The current version i have of mongoose lets me do this using the workaround described above. I dont see any way of performing this with the new version as you describe unfortunately.

cpq, 2020-12-23:
@MJLHThomassen what stops you from replacing fopen with ota_start, fwrite with ota_write, and fclose with ota_end? We have several OTA implementations using this mechanism, without spooling to flash.