cesanta/mongoose #1047: avoid aliasing standards function in public header file

azadkuh, 2019-08-25:
[`mongoose.h`](https://github.com/cesanta/mongoose/blob/d5beb7ba3f3767891f3d85945d7d33c1d8596e37/mongoose.h#L252) redefines some standard `c` functions by macro:
```c
#define snprintf _snprintf
#define vsnprintf _vsnprintf
// or
#define strdup _strdup
```

in different places.

such hacks in **public headers** are recipe for headache and conflicts with other libraries or user's code:
```cpp
#include <cstdio>
#include <mongoose.h>

int main(int argc, char** argv) {
    char test[32]  = {0};
    std::snprintf(test, 31, "%s/%s", "cesanta", "mongoose");
    std::printf("hello %s\n", test);
    return 0;
} 
```
fails to compile by `msvc` 2017/2015:
```
..\main.cpp(6): error C2039: '_snprintf': is not a member of 'std'
C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\stddef.h(19): note: see declaration of 'std'
..\main.cpp(6): warning C4996: '_snprintf': This function or variable may be unsafe. Consider using _snprintf_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\stdio.h(1961): note: see declaration of '_snprintf'
```

all of these hacks should go into `mongoose.c`.

cpq, 2020-12-23:
@azadkuh what other solution would you suggest?
Sprinkling all the code where `snprintf` is used, with `#ifdef _WIN32` ?

closed 2020-12-23 by cpq

azadkuh, 2020-12-23:
@cpq

IMHO there should not be any redefinition of standard library functions in **public** headers, the scope of such macros should be strictly limited to mongoose c files or private headers.

cpq, 2020-12-23:
Could agree with that, that makes sense.

On the other hand, this is not a redefinition. This is a definition.
Not sure what was the reason for MSVC developers not to define, for example , `snprintf` function - but it is not defined, as well as some other C99 constructs.

azadkuh, 2020-12-23:
I have little experience with MSVC (esp recent versions), but as the error code suggests, using `_snprintf`, ... are both invalid and unsafe, have a look at [godbolt.org](https://godbolt.org/z/sEYfrj)

> Would it be a solution to use `snprintf` instead of `std::snprintf`?

short answer: no, that smells

- in `c++`, using `std::xxx` is recommended over their c counterparts, and also including `<cname>` (instead of libc `<name.h>`) is a common/best practise
- we may also use other 3rdparty libraries with extensive use of `std::snprintf`, `vsnprintf`, ... (for example [nlohmann/json](https://github.com/nlohmann/json)). leaking these macros into the user's code causes numerous ugly compile errors
- using `snprintf` over `std::snprintf` doesn't help as the compiler emits the error for the definition, not the usage of this function

azadkuh, 2020-12-23:
as a quick hack (which also smells) you may add another macro check for `c++` as:
```cpp
#if defined(_WIN32) && !defined(__cplusplus)
#define snprintf _snprintf
#define vsnprintf _vsnprintf
#endif
```
this may work under `c` and also `c++`.

[godbolt.org](https://godbolt.org/z/8nzeab)

cpq, 2020-12-23:
That is a good suggestion, thanks.

Pushed https://github.com/cesanta/mongoose/commit/2cf1b097679d8b04926d034717dbf843d59b5f64