cesanta/mongoose #1050: connnecting to rest api https://akabab.github.io/superhero-api/api goes for infinite loop

srinivasan1995, 2019-09-19:
body of code :
static const char* url = "https://akabab.github.io/superhero-api/api";
static int s_exit_flag = 0;

static void ev_handler(struct mg_connection* c, int ev, void* p) {
	cout << "failure code = " << ev << endl;
	struct http_message* ev_data = (struct http_message*)p;
	switch (ev) {
	case MG_EV_CONNECT:
		if (*(int*)p != 0) {
			fprintf(stderr, "connect() failed: %s\n", strerror(*(int*)p));
			s_exit_flag = 1;
		}
		break;
	case MG_EV_ACCEPT:
		cout << " it is accpeted !!!!" << endl; break;
	case MG_EV_HTTP_REPLY:
		c->flags |= MG_F_SEND_AND_CLOSE;
		//if (s_show_headers) {
			fwrite(ev_data->message.p, 1, ev_data->message.len, stdout);
		//}
		putchar('\n');
		s_exit_flag = 1;
		break;
	case MG_EV_HTTP_CHUNK:
		fwrite(ev_data->body.p, 1, ev_data->body.len, stdout);
		s_exit_flag = 1;
		break;
	case 123:
		cout << "coolio dude " << endl;
		break;
	case MG_EV_CLOSE:
		if (s_exit_flag == 0) {
			printf("Server closed connection\n");
			s_exit_flag = 1;
		}
		break;
	default:
		break;
	}

	cout << "request handled quitting" << endl;
}

int main(void) {
	struct mg_mgr mgr;
	struct mg_connection* nc;

	mg_mgr_init(&mgr, NULL);
	mg_connect_http(&mgr, ev_handler, url, NULL, NULL);

	int count = 0;
	while (s_exit_flag == 0) {
		std::cout << "count = " << count << std::endl;
		mg_mgr_poll(&mgr, 1000);
		count++;
	}
	mg_mgr_free(&mgr);

	return 0;
}

i dont see action handler getting invoked !!!

i know i might be missing something . 

tried with below. 

	nc=mg_connect_http(&mgr, ev_handler, url, NULL, NULL);
	mg_set_protocol_http_websocket(nc);

i found that nc is null ?? 

could anyone help in resolving the issue ??( i guess it has probably got to do something with https)

nliviu, 2019-09-20:
You need to enable SSL.
Have a look at the example makefiles:
[http_client](https://github.com/cesanta/mongoose/blob/master/examples/http_client/Makefile)
[examples.mk](https://github.com/cesanta/mongoose/blob/master/examples/examples.mk)

srinivasan1995, 2019-09-21:
Hi ,
added openSSL and linked library , still i see this issue . any other's i am missing ?

```
cmake_minimum_required(VERSION 3.10)
set(CMAKE_CXX_STANDARD 11)

project(sslclient)

find_package(OpenSSL REQUIRED)

add_library(sslclient
include/main.hpp
include/mongoose.h
src/mongoose/
src/mongoose.c
)

target_include_directories(
 sslclient
 PUBLIC
   "include"
)

add_executable(final_out
 src/main.cpp
)

target_link_libraries(final_out sslclient OpenSSL::SSL)
```

nliviu, 2019-09-21:
[MG_ENABLE_SSL](https://github.com/cesanta/mongoose/blob/d5beb7ba3f3767891f3d85945d7d33c1d8596e37/examples/examples.mk#L15)

srinivasan1995, 2019-09-22:
thanks 👍  that did the trick

closed 2019-09-22 by srinivasan1995