cesanta/mongoose #1065: Segmentation fault by reading from address 0x0 in 6.16

dmundt, 2019-10-18:
MG_ENABLE_HTTP_URL_REWRITES is set and comma-separated list of `uri_pattern=url_file_or_directory_path` rewrites is used. If `uri_pattern` matches requested URI segmentation fault occurs. It seems `proto_data` field points to 0x0.

Behavior was introduced in Mongoose 6.15 and still exists in 6.16. Possibly related to #1019.

```
Program received signal SIGSEGV, Segmentation fault.
mg_http_reverse_proxy (nc=0x3312c8, hm=0x7efff2f0, mount=..., upstream=...)
    at D:/[...]/source/mongoose.c:8124
8124    D:/[...]/source/mongoose.c: No such file or directory.
(gdb) bt
#0  mg_http_reverse_proxy (nc=0x3312c8, hm=0x7efff2f0, mount=..., upstream=...)
    at D:/[...]/source/mongoose.c:8124
#1  0x00033d84 in mg_http_handle_forwarding (opts=<optimized out>, hm=<optimized out>, nc=<optimized out>)
    at D:/[...]/source/mongoose.c:8173
#2  mg_serve_http (nc=0x3312c8, hm=0x7efff2f0, opts=...)
    at D:/[...]/source/mongoose.c:8549
#3  0x00039494 in Service::OnRequest (nc=0x3312c8, ev=<optimized out>, ev_data=0x7efff2f0)
    at D:/[...]/source/service.cpp:653
#4  0x0002bf00 in mg_call (nc=0x3312c8,
    ev_handler=0x39198 <Service::OnRequest(mg_connection*, int, void*)>, user_data=<optimized out>, ev=0,
    ev_data=0x7efff2f0)
    at D:/[...]/source/mongoose.c:2462
#5  0x0002d47c in mg_http_call_endpoint_handler (nc=0x3312c8, ev=100, hm=0x7efff2f0)
    at D:/[...]/source/mongoose.c:8947
#6  0x0002e1fc in mg_http_handler (nc=0x3312c8, ev=<optimized out>, ev_data=<optimized out>)
    at D:/[...]/source/mongoose.c:6753
#7  0x0002bf00 in mg_call (nc=0x3312c8, ev_handler=0x2def0 <mg_http_handler>, user_data=<optimized out>,
    ev=0, ev_data=0x7efff6e8)
    at D:/[...]/source/mongoose.c:2462
#8  0x00030a40 in mg_recv_tcp (len=<optimized out>, buf=<optimized out>, nc=<optimized out>)
    at D:/[...]/source/mongoose.c:3010
#9  mg_do_recv (nc=0x3312c8)
    at D:/[...]/source/mongoose.c:2965
#10 0x000312b8 in mg_if_poll (nc=0x3312c8, now=1571386709.505996)
    at D:/[...]/source/mongoose.c:2519
#11 0x000315d8 in mg_mgr_handle_conn (nc=0x3312c8, fd_flags=1, now=1571386709.505996)
    at D:/[...]/source/mongoose.c:4059
#12 0x00031bd0 in mg_socket_if_poll (iface=<optimized out>, timeout_ms=<optimized out>)
    at D:/[...]/source/mongoose.c:4290
#13 0x00020da8 in mg_mgr_poll (m=0x314220, timeout_ms=1000)
    at D:/[...]/source/mongoose.c:2667
(gdb) p be->proto_data
$1 = (void *) 0x0
(gdb)
```

dmundt, 2019-10-18:
Provided a fix for the issue above with PR #1066

cpq, 2020-12-25:
Please use the latest codebase and example at https://github.com/cesanta/mongoose/tree/master/examples/http-reverse-proxy

closed 2020-12-25 by cpq