cesanta/mongoose #1070: [Feature request]: MQTT 5

krombel, 2019-10-24:
To use MQTT 5 I was asked to create an issue here.

More details you can find in the mqtt-libs repo: https://github.com/mongoose-os-libs/mqtt/issues/12

cpq, 2019-10-29:
Thank you.

Could you elaborate on your specific use case please - what project you're working on?
Or this request does not have an underlying project, but rather is a general upgrade request?

cpq, 2020-12-17:
No response, closing this.

closed 2020-12-17 by cpq

krombel, 2021-02-16:
Sorry for the late response.
Regarding your questions: I was in the progress of working on an IoT-Project which involved pushing metrics via MQTT.
I wanted to run latest tech on that to not be in the position of running a migration in future.

In the meantime we went for MQTT 3.1.1 leaving this as general upgrade request.