cesanta/mongoose #1071: mongoose udp mg_send error

MarkMaYJ, 2019-11-01:
hi,
   I have read the mongoose souce code and I have found a problem on the function mg_send() when sendｉng udp data.
   when call mg_send() twice or more, mbuf is byte based buffer, mg_send() just append data to the buffer, and the socket sendto function just use the full mbuf data

nliviu, 2019-11-01:
That's the way it is intended to work: https://github.com/cesanta/mongoose/blob/dce60c6dbb096f3b96e1a45cbfdfd55e18b38bb6/mongoose.h#L4377-L4384

rojer, 2019-11-01:
i'd say, it's a known issue. there is currently no way to send separate UDP datagrams in the same poll cycle, and there should be.

cpq, 2020-12-24:
The latest codebase does not spool data to the iobuf for UDP send, instead, it calls `sendto` directly.
The receiving codepath also triggers `MG_EV_RECV` for each UDP datagram - and spools it to the `c->recv` iobuf.
It is up to the application to cleanup `c->recv` after each datagram, or to keep it.

closed 2020-12-24 by cpq