cesanta/mongoose #1073: mongoose parse Transfer-Encoding: chunked type  http sever response err!!

jyl58, 2019-11-09:
Version: 6.15, 6.16
Hi,all:
When I parse the response from http sever I meet a problm. I get follow response at event MG_EV_HTTP_REPLY call back, but the content is not correct. after again and agian test , I find there is some bad string in the body.  I sign it use bold font, and it is the size of next chunk . so anyone know how to solve it.
==============================
HTTP/1.1 200 OK
Server: nginx
Date: Sat, 09 Nov 2019 04:45:32 GMT
Content-Type: text/json;charset=UTF-8
Transfer-Encoding: chunked
Connection: close

2liHtFH1efR3lajnXhj6kK7GTwD2t8OzkfysQwoKAq1/r2/BQsZa7PkSi1oWQi7tPmnACZ7pVizLuvob7kupc7JhsgCmj1kFuCEmCmSwXCxjcRBJ664O0D5f4AOSgFzcJfQM/b3V0sp+2e6ZoEtzyp8qBD7yNlQys90BJ5FmW+xxoo9lQ/kK7o82Gzbu7Vp6t+ezGxqkPeZeURuW8v+2W4ZczDcV1MEtgTznBnQEgjMlcdNkfXOWUeD8WH1Ifwbd2McylYlFv5uvcLa/x0QYZd3/ddso3dH4RxpmIRjMiC7U3IQkXEFjEWK9ZS7a6r3nm851zqfi1paSizTpNkoYfgj0CoIJrlr5llt4S3LnwXcy8CJktHmP7fjxjvg6mSWSdYN9ht9XJ7fC5y5PsBdvjNmoZcBqmEFk7cguesLDfPdNuJDvvGvjpmPe1glnYn1YMyp/A2Ulcz1QSVFHba5S8sPS3ElTmb86dkScVUGvmvIsDFpYKAL0RI6iPFkdED7sj/4MW7BiXGe6O15qOJce+mDp6kuToOdQtu083zCeFkNZXPmuzSx1QO1ECvprvczDIq2JEau7Az6gHqnrxuseHsCfzOREdFuwUIIg1dendR68IXxYiJUhqwhD9babRZSer0vBWmdQrh/1hWfMvruB2tcEBKrqc+pcJMW+BInRGAIYYsV4kk8THgdO934pKMIDoqg/JzZ6KIsiC8HaiXXIjnSg/9R1HO3yXIoQRi+ipzDtiqxq2Hs1QjtxbLm0pKILpVwHtyW1CxfkEQYQysY1DLSDyIt4SPRyESFZy7Utt7ZvGBUB1ytLwWzXYAR3T+3xgL/vTZstubfa6AtxRtGjx4FyVEKfRNUP6jQ5bcEBTb10W2ZOVqG291MK1KcgL/cILaVqBOd3tje4ho4kMiVTI2bGgVG7R7cO9etfFMp3oPffiOgWXuvEDUw3j2LjStCQULTy0UnANkzPIhOz6dJrlHekb6Kj6SLBynC9v9guYYAoDZPpp1th5C/wBQCE814XG2Ev7l4r9ww726ViAsEbcKY+SMbzMGyAmXev2KhK9cEsZdSsIdWkiOJGa9QqUtqQaM4iraJRwUHBet/ycL1VubQdXdSgCPNiA5vryFLf8TcHWi2lNxkzk7Urtp+HtdpCh6yJhmn+79qiMN/LaAfYkN+4zkdaYgcgRT6HF2pkxTUF1IpjAPEEyQlDOJShrLwtIYBHYDWi8SduNpy13cpxbdXA5S2xSjf4QFOVi+AnddmzmUV9Nn8E6ikDpDdcY//X/LJMpJVva4Cff72Gln1SvlqdgWzGBflgPG/dFpbErefj6bMC97WM8oSmErOQd8aBLOV9siAJo0OduntVasV/94TLFcjhsKnVq7HqLjtTVlx0JsI6XZxHKf1oVbTp/hw5caKPZUP5Cu6PNhs27u1aeimKGSGy2ZT8MPwHi4PM2zhm5jBdUkjlcSvODKjUvqzBmEFnXNlkwqyZhrvNKvNHVdq8heVBw9i/erDZ7DA8lDqfXHdWh46KKV9VXdW1Urb25Ya4yiOJKWzbK43GbUwx6zYAwH83TjpOAb6u059H/mi/XKlpbKQA1iRjT4/P+6v36y0uLBKk7+Tf+kzC+cEY0W5s70Re9GU199J1yJX0Wq7XvFv8WcBM+CSmeBF7emGWhWMqgRN3bdpgCkLz2teUQbKPmPU5GaCyMN563EDUu7L/3HRpM93xibz3V0tdiZd6BFGwO8Dxd0lF62fL+Y0PEDTzISD07LxL5NufDS4qM+i2cQsw4zxCIbtP1RaqmkPjjKTZKgqgU6KH7BfUCjoj59DGhg5xDcDtyx7ZB5/Bv1W5G2oAYqbEICOGiTG2KLb0Yzo42Hzc/0qsq2tbuGyVajDNnmhBW2o1yPkzSqf21WvhA6hIW8OEre3hMG5PBXrgER1RuPxQAAp4ReI4yw1WwoZ+h0fTrr/usSpvSHugKkgU35AyiIa3frZ/MQt3YWZ1TUIWn69iA9QZZ46Wc1QQPF2hz9VxcA7bYR1eFYaGKvw1EilzT3M9Cx39rDCgdBJh9QaonF+P8lYPZcC0V4SpkRV2V8g4QMJwceHka4F/DU5LYPihM087tVXRkoUgjsLgZSHt3W71qymZEGewUHLcC7xWWIF45Zyvsu35nCGSgMOjnAYQaJ01EVo/JQ/ge7OXK+48vpXUyUegsA+SmWooDEian3616U6MU0Z+qNVro5Aaha6m1kq19xSnz9teNlAD5kj06+f6x92TNW0etn/pnv5rsqwDEfEFDPUJ4ZbpkdPBEpXoe0KJ3IoL+wTIemGszD8uTtZFzoUaErKNjPskZG6k9J8EIhhQIEOVaLts/FMvecWIGExJnIbFMpzlnDNRGZnCNPLcHrl2vnMlZPHdONTchCRcQWMRYr1lLtrqveebznXOp+LWlpKLNOk2Shh+CPQKggmuWvmWW3hLcufBdxJ83ztl/tC0T4y18398ZmV1g32G31cnt8LnLk+wF2+M2ahlwGqYQWTtyC56wsN894n1r+XTGirY2WUCrVce0hh0S/aMkHGM7TQQSBQnn475uK/JLr4uTAScRlQ4PfE3X0tfxwyhIjHGKjKKoEVuX0y1rmVEdGqqTYT17pfrPWnsMipqhMfESibbYkydXK8hr3bNKtT9EvzhzWJRBbW+SXviJwK8WHiknOFYymGiq1zJNSa0NxydPMKTLUjqyt2/E0sRFbSkyrMZl3IapUj+CnOvS8FaZ1CuH/WFZ8y+u4Ha9Al9VI84flQNX4L3Uemy+QfIFK3QOJJP487a75zvToK0HV3UoAjzYgOb68hS3/E3B1otpTcZM5O1K7afh7XaQoesiYZp/u/aojDfy2gH2JDObxrFs2GA+eIny7SpbrzWcylVBhITJBXea0xD6TVvXSGAR2A1ovEnbjactd3KcW2mEwNXEe8FI+ul46DNZR65YT6g5uSi5s0kUh4z/W/lNeHvNI5vV3ZFp3OVM4ya6RLeCGHLTk5Jcr//np6UDq4veyjOHV9QcyFA77F3km4/YLl1EPEygRt7sX5Ic2xpLpr/6edcBv46FLkXeJVUekG4qjVWCjY1WNwP1OfRxhbTxXGij2VD+QrujzYbNu7tWnrrbVc+fkafpEqhoM9SPaMNhlzMNxXUwS2BPOcGdASCM9qdf/NUeilZ7dFl6OvczCLHKe1BjZNAsd1+YNsWXbKZ5Z5H2NP1ZN85KB8FZwQphBvG5Xkmh6NmGg8ai5FS5Q5m5py2ysPm5ihHpnul9u/EwI/zttAjhlnaigjgRw2A0UCc7EXaUtoDRXUSBX48aiOuQwZl655M2p4aMBe7+LoTeDogEKfAge6l9TH/n+FBL58scGPkgjzGjtqzjKsOmxJD4suCmg/uY7McciU6e5m9rm+t/+0taw/LasGZgx/tyR6WTtSpSIwVflrv2UTfoF8ihY1AR31IyKH1OLY2fLPbd4t+HxuOBra7Li+BECQRHm2fNPbfS3oNJ2QPh/ugE3gVwSty6wxvk4WkdD/9TDg28UQGSuN+x6rlXQhYb3kYGIlsZuk78mY4pPUok+7o9teLIOj5XP5TU1WsknrUrVg3FnzIQP95wpNTsn3WDerHAIBebu8F0XCJ1tLUYzeL3AQXjIsywheEUd26C0Syi2vqFN+QMoiGt362fzELd2FmdU1CFp+vYgPUGWeOlnNUEDxdoc/VcXAO22EdXhWGhir8Jfg9xx338UyffxeeUwHkwR/jJZ3OgSqXUh+thtTrwgMVdlfIOEDCcHHh5GuBfw1O49TlTuRVBp4EUtBv1k0iibpf8xqSxplTBtNYTpW96GnY9kSMs3cdVAvulEm7ClLlyNXbylEMtJoGSRaGXKFx4BpomTg3S9lGNl7XHDqU2s5+T1QOZ1D5cwB0DjNlC1w9EGcpCtsrWLInUw5pViReC5LKruNb3M30Ncny/QX3/iaCLXjbwwN**1fa**
L/iF/HE+5/OF4FW6qTJaM/Okr/8TRN1uYZ8w/Lk7WRc6FGhKyjYz7JGSKE6f7vJWV6WLq0SGR6r1aUHZl/MldtrGzZCK6A1T9widxZV8Slwi0mTPrK/Vux1T6DLDQr6jIoxDv/296Y6V8j3V8TmnHE36wpF2IHrw6UTpVEkatJoWDLUjARqA0TLyW4XmT9IQij1H5OslVGvgMDCayPxc1aYCtrEBTkahU813EyP55gt2NhUzxFvmjyVpiM/sILNaW77fOogSQ6rP7UFwrNfxzaH/8rBRI/vMb13Luxm8GJmzgwl1imK8Jz10SM2b5d2+VDvbRf2CgY0r5TR0+qrDSl3yqc0Z+qBlYlNsWREmEwXfJ5fiyVd8oOEQ8+snrrR2BV0kYJPTLE95C1zFZO6cJ5I5H8WiBn/uNqXz7j0qqhnJK1Lr9U8WLENDO83j9wUpCCU/LMenVlLWdteHySaupjkb26Fo4pKSKOe8ZzcUc+KwRFeFXSCVmepI=

0

jyl58, 2019-11-09:
I use it on linux embedded system.

jyl58, 2019-11-09:
@cpq  Hi,Sergey , could you view this problem.

cpq, 2020-12-25:
Please use the current codebase.
As of now, it does not process chunked data.

closed 2020-12-25 by cpq