cesanta/mongoose #1078: simplest_web_server_ssl compiles with error: statement with no effect.

schilton, 2019-11-20:
When I try to compile the example program, I receive the following error below.  As a newbie, I would really appreciate an explanation of what is going on and how I should resolve this.  If it has "no effect", presumably I could just comment it out.  However, I realize that since the source has been viewed my many and there is not an issue here or in the forum, it is probably my problem.  So thank you to anyone taking the time to educate me.

In file included from ../../mongoose.c:4670:0:
../../mongoose.c: In function ‘mg_use_cert’:
/usr/local/ssl/include/openssl/ssl.h:1486:52: error: statement with no effect [-Werror=unused-value]
 #  define SSL_CTX_set_ecdh_auto(dummy, onoff)      ((onoff) != 0)
                                                    ^
../../mongoose.c:4958:5: note: in expansion of macro ‘SSL_CTX_set_ecdh_auto’
     SSL_CTX_set_ecdh_auto(ctx, 1);
     ^~~~~~~~~~~~~~~~~~~~~

nliviu, 2019-11-20:
This is not related to mongoose, but to your openssl.
Try to add `-Wno-unused-values` in [examples.mk](https://github.com/cesanta/mongoose/blob/dce60c6dbb096f3b96e1a45cbfdfd55e18b38bb6/examples/examples.mk#L2) `CFLAGS` line, or remove `-Werror` in the same line, or update your openssl library.

cpq, 2020-12-24:
Thanks @nliviu 
Closing this, as there is no response so far.

closed 2020-12-24 by cpq