cesanta/mongoose #1080: Mongoose task Freeze when compiling with GCC 8.2.0

StefanRvO, 2019-11-28:
Hi.
I'm using the mongoose webserver for my project which uses an ESP32.
I recently tried upgrading to GCC 8.2.0 from 5.2.0, but ran into trouble with the webserver. Basically, a few ms after a client connects to the server, my task with handles the webserver will stop running.
The relevant code can be found here: https://gitlab.com/StefanRvO/light-equalizer-platform/blob/master/software/firmware/main/modules/web_server_module/web_server_module.c

I'm quite sure it is not a problem with my code, as event if i comment out all code in my event handler, the problem still occurs. Basically, the while(1) loop will get stuck. I have not been able to pinpoint exactly where code execution gets stuck, but it is somewhere inside the mg_mgr_poll() call. Everything works fine with GCC 5.2.0.

Any ideas what might be the issue?

Everything works fine

nliviu, 2019-11-28:
Could you provide a simple project to reproduce the isuue, please?

StefanRvO, 2019-11-30:
I have made this example which demonstrates the issue with as little code as i could:
https://gitlab.com/StefanRvO/mongoose_8_2_0_issue/blob/master/main/main.c
When compiling with 5.2.0, the "Webserver loop" print continues to happen after creating a connection and sending some data, (e.g. using telnet), while it will stop being printed if compiled using 8.2.0 just after connecting and sending some data.

I'm usingthe newest esp-idf master(93a8603c545fb8e54741d6685146e2f3b874378d).

with these GCC versions:
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0

To get my above example running, you need to change the wifi SSID and password in the top of the main file to some appropriate ones, and the "CFLAGS+=-Wno-error=format-truncation" in the makefile needs to be commented out to compile with gcc 5.2.0.

I will see if i can reproduce the issue on x86, as it may be easier to pinpoint the issue if it can be reproduced locally instead of on the ESP32.

StefanRvO, 2019-11-30:
Can't seem to reproduce the issue with GCC 8.3.0 on linux x86_64 (couldn't easily download 8.2.0), so it may be something strange going on with the xtensa compiler.

closed 2019-11-30 by nliviu