cesanta/mongoose #1081: License: GPL-2 vs Apache

sbabic, 2019-11-29:
Hi,

even if mongoose is licensed under GPLv2, code in mongoos.c contains headers from Apache-LICENSE-2.0. The two licenses are declared incompatible by FSF. This was introduced by commit
b8eca17c558525cea5e1e81594897021eef239dd in mongoose.c. I just ask if this is a license issue - should the Apache header be changed with a GPL-v2 compliant header ?

anatolyborodin, 2019-12-13:
https://www.apache.org/licenses/GPL-compatibility.html : "APL & GPL3 - compatible, APL & **GPL2**(!!!) - incompatible"

sbabic, 2019-12-13:
Right, but mongoose is licensed under GPL-V2...

bgermann, 2020-02-24:
This can be closed because those license notices were changed.

cpq, 2020-12-24:
The aforementioned commit was a mistake: mongoose sources got that license snippet because other software was processed in batch, and mongoose sources were erroneously affected.

These snippets were deleted.
There are no Apache-licensed code in Mongoose library.
Closing this.

closed 2020-12-24 by cpq