cesanta/mongoose #1084: http download big file

Gaoy9303, 2019-12-13:
If the browser is a client, is there an example of HTTP client downloading a large file from the server writed by mongoose? In the example, only the example of uploading a large file is found.

RiccardoDalFiume, 2021-01-27:
You can serve it using mg_http_serve_file, although you need filesystem support.
If you need to serve file from ROM, in mongoose 6.18 the only way I found was to reimplement mg_fopen/mg_fread and redefine close() via macro, since it wasn't available as a mg_* version.

I haven't investigated a way to perform this in 7.x since much of the API has changed and still subsceptible to changes.
Maybe @cpq has insight for this?

cpq, 2021-01-27:
Yes, https://cesanta.com/docs/#mg_http_serve_file  is a friend here.

Serving a "file" from memory could be done via a single call to `mg_printf()` if a "file" fits into RAM, or a more elaborate state machine handler should be implemented that limits the transfer and sends data in small chunks. See `mg_http_serve_file()` implementation in `src/http.c`.

RiccardoDalFiume, 2021-01-27:
If I understand correctly I should look for MG_EV_WRITE event?
is there an example to clarify usage?

cpq, 2021-01-27:
@RiccardoDalFiume as I have mentioned, an example is serving static file, in http.c

The logic is simple:

1. mg_http_serve_file opens a file, and allocates a state strucuture  in the connection's data, `c->fn_data`. A state structure (`struct http_data`) contains an opened `FILE *`
2. Then mg_http_serve_file changes the event handler to a special event handler function:
```
        d->fp = fp;
        d->old_pfn_data = c->pfn_data;
        c->pfn = static_cb;
        c->pfn_data = d;
```
3. From that point on, a `static_cb` will be called, which is a very simple function that catches MG_EV_WRITE to see whether output buffer is drained enough to send more data. When connection is closed  or all data is sent, an old event handler is restored.

closed 2021-01-27 by cpq