cesanta/mongoose #1085: mg_broadcast() is not thread-safe

dd1dd1, 2019-12-15:
I am updating the web server component of our data acquisition application from mongoose 6.4-ish to mongoose-6.16 (https://midas.triumf.ca).

By necessity, our web server is multithreaded - we have many "web requests" that take a very long time to process (databases have to be queried, large data files have to be read, etc) - and it took me a few years to work up the courage to tackle the new multithreading model implemented in recent versions of mongoose.

I generally like how the removal of old multithreading code simplified the mongoose code and api.

But I have run into trouble working with the new code:
(a) examples/multithreaded/multithreaded.c does not explain how to send arbitrary large amounts of data (more than fits in MG_CTL_MSG_MESSAGE_SIZE).
(b) contrary to https://github.com/cesanta/mongoose/issues/1068, mg_broadcast() is obviously not thread-safe. True, it is safe against mg_mgr_poll() & co, but not safe against multiple threads calling mg_broadcast().

Ultimately I got my web server working, with two changes to mongoose:
1) mg_broadcast() has to be protected by a mutex (in user code)
2) mg_mgr_handle_ctl_sock() should send the handshake reply after the loop over network connections (required change in mongoose.c).

This is how I explain the problems to the next guy to work on our web server:

   // NB: mg_broadcast() is advertised as thread-safe, but it is not.
   //
   // in mongoose 6.16, mg_brodacast() and mg_mgr_handle_ctl_sock() have several problems:
   //
   // a) "wrong thread" read from mgr->ctl[0], defeating the handshake
   //
   // b) "lost messages". if more than one message is written to mgr->ctl[0], the second message
   //    will be "eaten" by mg_mgr_handle_ctl_sock() because of mistatch between number of bytes read and written
   //    in the two functions. mg_mgr_handle_ctl_sock() always reads about 8000 bytes while mg_broadcast()
   //    writes 8 bytes per message, (per examples/multithreaded/multithreaded.c. mhttpd messages are a bit longer).
   //    So if multiple messages are present in the msg->ctl[0] pipe, the read call (of about 8000 bytes)
   //    in mg_mgr_handle_ctl_sock() will return several messages (last message may be truncated)
   //    but only the first message will be processed by the code. any additional messages are ignored.
   //
   // Problems (a) and (b) are easy to fix by using a mutex to serialize mg_broadcast().
   //
   // c) if the mg_broadcast() message contains pointers to the data buffer to be sent out,
   //    the caller of mg_broadcast() should not free these data buffers until mg_send() is called
   //    in "on_work_complete()". In theory, the caller of mg_broadcast() could wait until on_work_complete()
   //    sets a "done" flag. In practice, if the corresponding network connection is closed before
   //    mg_mgr_handle_ctl_sock() has looped over it, on_work_complete() will never run
   //    and the "done" flag will never be set. (Of course, network connections are permitted to close
   //    at any time without warning, but) the firefox browser closes the network connections "a lot"
   //    especially when user pressed the "page reload" button at the moment when HTTP transations
   //    are "in flight". (google-chrome tends to permit these "lame duck" transactions to complete and mongoose
   //    does not see unexpected socket closures, at least not as many).
   //
   // To fix problem (c) I need to know when mg_mgr_handle_ctl_sock()'s loop over network connections
   // has completed (two cases: (a) my on_work_complete() was hopefully called and finished,
   // and (b) the "right" network connection was already closed (for whatever reason) and my on_work_complete()
   // was never called).
   //
   // My solution is to change the handshake between mg_broadcast() and mg_mgr_handle_ctl_sock() by sending
   // the handshake reply after looping over the network connections instead of after reading the message
   // from msg->ctl[1].
   //
   // This requires a modification to the code in mongoose.c. If this change is lost/undone, nothing will work.
   //

K.O.

dd1dd1, 2019-12-15:
This is how I call mg_broadcast() -

   s_mg_broadcast_mutex.lock();
   mg_broadcast(&s_mgr, on_work_complete, (void *)&res, sizeof(res));
   s_mg_broadcast_mutex.unlock();

This is the diff of my mongoose.c against git tag 6.16 -

***************
*** 4114,4122 ****
    struct ctl_msg ctl_msg;
    int len =
        (int) MG_RECV_FUNC(mgr->ctl[1], (char *) &ctl_msg, sizeof(ctl_msg), 0);
-   size_t dummy = MG_SEND_FUNC(mgr->ctl[1], ctl_msg.message, 1, 0);
    DBG(("read %d from ctl socket", len));
-   (void) dummy; /* https://gcc.gnu.org/bugzilla/show_bug.cgi?id=25509 */
    if (len >= (int) sizeof(ctl_msg.callback) && ctl_msg.callback != NULL) {
      struct mg_connection *nc;
      for (nc = mg_next(mgr, NULL); nc != NULL; nc = mg_next(mgr, nc)) {
--- 4114,4120 ----
***************
*** 4124,4129 ****
--- 4122,4131 ----
                         ctl_msg.message MG_UD_ARG(nc->user_data));
      }
    }
+   // change from original mongoose code: send the reply message
+   // after we finished processing the broadcast message. K.O.
+   size_t dummy = MG_SEND_FUNC(mgr->ctl[1], ctl_msg.message, 1, 0);
+   (void) dummy; /* https://gcc.gnu.org/bugzilla/show_bug.cgi?id=25509 */
  }
  #endif

dd1dd1, 2019-12-15:
With these changes, both problems are fixed: "eaten messages" and "I need to know when on_work_complete() finishes, even if it was never called".

Please commit my change to mongoose.c and document the potential need for protecting mg_broadcast() with mutex in examples/multithreaded/multithreaded.c. (as-is, this toy web server does not need a mutex).

Thank you for the good work on mongoose, best of wishes and a happy new year!

K.O.

kokole, 2020-03-15:
I've been getting from memory to deadlocks trying to implement correct multi threading in mongoose.
This is very good information.

Yogurt4, 2020-09-14:
Thank you for this fix.
I hope someone will merge it into the main source, being such a fundamental functionality, and with you serving the solution.

MalcolmBoura, 2020-11-08:
I haven't looked at the source but it sounds to me as if this function blocks until the main thread has completed the traversal of the connections. One of the reasons for using a thread is to ensure that processing is not overly delayed by activity in another thread. Hence blocking APIs that wait on another thread should be avoided.

The documentation makes no mention of data lifecycle and it does need to. For example is the data copied? When can it be freed? Who frees it?

I suggest that the handler func is called an additional time after the connection traversal with a NULL connection pointer to indicate that the data passed and any associated data structures, can safely be deleted.

There should be no need for the different thread requirement.

I am surprised that this function does not use a thread safe queue. Faster, less memory, insignificant executable size.

cpq, 2020-12-23:
The `mg_broadcast` is gone - multi-threaded support is refactored using socketpairs.
Please see an example at https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded

closed 2020-12-23 by cpq