cesanta/mongoose #1087: mg_http_reverse_proxy() needs fixes

dd1dd1, 2019-12-21:
In mongoose 6.16, mg_http_reverse_proxy() does not work without following fixes.
a) after mg_connect_http_base(), mg_http_get_proto_data(be) returns NULL and there is an immediate crash
b) purl is missing the query_string
c) mg_http_handler() overwrites our proto_data, after that, our reverse_proxy_data.linked_conn becomes NULL, in mg_reverse_proxy_handler(), pd->...linked_conn is NULL and nothing works.

With following changes against mongoose.c git tag 6.16, http proxy works:
<pre>
diff --git a/progs/mongoose616.cxx b/progs/mongoose616.cxx
index cc33e4fe..781b3559 100644
--- a/progs/mongoose616.cxx
+++ b/progs/mongoose616.cxx
@@ -6651,7 +6651,9 @@ void mg_http_handler(struct mg_connection *nc, int ev,
 
     if (req_len > 0) {
       /* New request - new proto data */
-      pd = mg_http_create_proto_data(nc);
+      if (!pd) {
+        pd = mg_http_create_proto_data(nc);
+      }
       pd->rcvd = io->len;
     }
 
@@ -8107,8 +8109,8 @@ void mg_http_reverse_proxy(struct mg_connection *nc,
   memset(&opts, 0, sizeof(opts));
   opts.error_string = &error;
 
-  mg_asprintf(&purl, sizeof(burl), "%.*s%.*s", (int) upstream.len, upstream.p,
-              (int) (hm->uri.len - mount.len), hm->uri.p + mount.len);
+  mg_asprintf(&purl, sizeof(burl), "%.*s%.*s%s%.*s", (int) upstream.len, upstream.p,
+    (int) (hm->uri.len - mount.len), hm->uri.p + mount.len, (hm->query_string.len > 0 ? "?" : ""), (int) hm->query_string.len, hm->query_string.p);
 
   be = mg_connect_http_base(nc->mgr, MG_CB(mg_reverse_proxy_handler, NULL),
                             opts, "http", NULL, "https", NULL, purl, &path,
@@ -8122,6 +8124,8 @@ void mg_http_reverse_proxy(struct mg_connection *nc,
     goto cleanup;
   }
 
+  mg_http_create_proto_data(be);
+
   /* link connections to each other, they must live and die together */
   mg_http_get_proto_data(be)->reverse_proxy_data.linked_conn = nc;
   mg_http_get_proto_data(nc)->reverse_proxy_data.linked_conn = be;
</pre>

K.O.

dd1dd1, 2019-12-21:
I invoke the proxy from my handle_http_message(nc, msg) like this:
<pre>
   std::string uri_encoded = mgstr(&msg->uri);
   std::string uri = UrlDecode(uri_encoded.c_str());
...
   if (starts_with(uri, "/proxy")) {
      mg_str mount = mg_mk_str("/proxy");
      mg_str upstream = mg_mk_str("http://localhost:8081");
      mg_http_reverse_proxy(nc, msg, mount, upstream);
      return;
   }
</pre>

Again, thank you for the good work on implementing the http proxy function in mongoose. Happy New Year!

K.O.

cpq, 2020-12-25:
Please use the latest codebase and example at https://github.com/cesanta/mongoose/tree/master/examples/http-reverse-proxy

closed 2020-12-25 by cpq