cesanta/mongoose #1095: Question about loop in the multithread example

KingDuckZ, 2020-01-21:
In the multithread sample code, inside on_work_complete() and right after the line "struct work_result *res = (struct work_result *)ev_data;" I added this line of code:

```c
printf("user_data=%lu, res.conn_id=%lu, nc=%p, c=%p\n", (unsigned long)c->user_data, res->conn_id, nc, c);
```

then I run the program, access 127.0.0.1:8000 from firefox and I get this output on the console:

```
Started on port 8000
user_data=1, res.conn_id=1, nc=0x55fa77005d70, c=0x55fa77005d70
user_data=1, res.conn_id=1, nc=0x55fa770052e0, c=0x55fa77005d70
```
Could someone explain what's going on? Is it normal that this code runs twice? This means mg_send_head() is also called twice, and I don't quite understand who's receiving the second response. Any further explanation is much appreciated, thanks.

warmheart21, 2020-04-13:
Dear Sir,

Did you fix this issue?
I have found the same log when using mongoose in multi-thread mode.
I traced my program step by step, and found the callback on_work_complete will be called in 
the mg_mgr_handle_ctl_sock. The ctl_msg.callback is the on_work_complete:
_**static void mg_mgr_handle_ctl_sock(struct mg_mgr *mgr) {
  struct ctl_msg ctl_msg;
  int len = (int) MG_RECV_FUNC(mgr->ctl[1], (char *) &ctl_msg, sizeof(ctl_msg), 0);
  size_t dummy = MG_SEND_FUNC(mgr->ctl[1], ctl_msg.message, 1, 0);
  DBG(("read %d from ctl socket", len));
  (void) dummy; /* https://gcc.gnu.org/bugzilla/show_bug.cgi?id=25509 */
  if (len >= (int) sizeof(ctl_msg.callback) && ctl_msg.callback != NULL) {
    struct mg_connection *nc;
    for (nc = mg_next(mgr, NULL); nc != NULL; nc = mg_next(mgr, nc)) {
      ctl_msg.callback(nc, MG_EV_POLL,
                       ctl_msg.message MG_UD_ARG(nc->user_data));
    }
  }
}**_
The mg_mgr_handle_ctl_sock use one loop to check all connections in the connections list of **_mgr_**, and the on_work_complete also has one loop. That's dual loop.
But one question is that the mgr pointer of every connection in the list all point to the same 
**_mgr_** . That is to say no matter which connection transfer into the on_work_complete, 
the first connetion in the loop is the same active_connection of the **_mgr_**.
Can we  remove the loop of the on_work_complete?
Thanks a lot for any comment!

cpq, 2020-12-24:
The multithreading support has been changed since 7.0 release.
Please see https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded

closed 2020-12-24 by cpq