cesanta/mongoose #1097: MG_ENABLE_IPV6 breaks usage of nc->user_data

jcorporation, 2020-02-11:
When compiling monoogose with MG_ENABLE_IPV6 the custom set ``nc->user_data`` is not usable.

```
static void ev_handler(struct mg_connection *nc, int ev, void *ev_data) {
    ....
    switch(ev) {
        case MG_EV_ACCEPT: {
            nc->user_data = (void *)(intptr_t)mg_user_data->conn_id;
            // here is nc->user_data 
            LOG_DEBUG("New connection id %d", (intptr_t)nc->user_data); correct
            break;
        }
        case MG_EV_HTTP_REQUEST: {
            // here is nc->user_data broken
            LOG_DEBUG("HTTP request (%d): %.*s", (intptr_t)nc->user_data, (int)hm->uri.len, hm->uri.p); 
```

jcorporation, 2020-06-30:
Sorry, was an error on my end

closed 2020-06-30 by jcorporation

PerryWerneck, 2020-06-30:
Which error? I'm having the same issue.

jcorporation, 2020-07-01:
I have not declared the MG_* flags for source files that include mongoose.h