cesanta/mongoose #1099: OpenSSL license exception

bgermann, 2020-02-24:
Please consider adding an OpenSSL license exception to mongoose's license.
There are 2 reasons for it.
1) [The GPL2 and OpenSSL license are incompatible](https://people.gnome.org/~markmc/openssl-and-the-gpl.html). So distributing a binary file with mongoose and OpenSSL included/linked is legally problematic on systems where OpenSSL is not supposed to be an integral part of the operating system. E.g. [Debian takes the position that this is not allowed](https://ftp-master.debian.org/REJECT-FAQ.html).
2) OpenSSL 3.0 will be Apache v2 licensed. The GPL2 and Apache v2 are incompatible as already pointed out in #1081. This can also be addressed by adding an "or later" clause to the license because Apache v2 and GPL3 are compatible.

bgermann, 2020-02-25:
This was [previously discussed](https://gitter.im/cesanta/mongoose/archives/2015/04/16) but I cannot see any outcome in the source file.

cpq, 2020-12-23:
@bgermann could you provide an example of a project that implemented it, please?

bgermann, 2020-12-23:
You can have a look at the deluge project.

cpq, 2020-12-23:
Per https://git.deluge-torrent.org/deluge/tree/LICENSE,
that's a GPLv3, which is a totally different license, unless I am missing something.

bgermann, 2020-12-23:
Yes, but the same exception can be used for GPL2

cpq, 2020-12-23:
Thanks for the background!
But no, we won't do that.
For those who care, a commercial license is available, which does not have GPLv2 restrictions.

closed 2020-12-23 by cpq