cesanta/mongoose #1102: if server as file server,  when  download txt file, it will open it directly

jiephe, 2020-03-11:
my code is like this:
`		mg_mgr_init(&m_mgr, this);
		mg_connection *connection = mg_bind(&m_mgr, m_port.c_str(), OnHttpEvent);

		mg_set_protocol_http_websocket(connection);
		memset(&s_http_server_opts_, 0x0, sizeof(s_http_server_opts_));
		// Serve current directory
		s_http_server_opts_.document_root = ".";  
		s_http_server_opts_.enable_directory_listing = "yes";	
		s_http_server_opts_.custom_mime_types = ".log=text/plain;";	
		while (true)
			mg_mgr_poll(&m_mgr, 500);
`

if server as file server,  when  download txt file, it will open it directly

agent-etienne, 2020-03-12:
This is normal. Every server does this.

jiephe, 2020-03-13:
> This is normal. Every server does this.

if i want browser download the txt file, i need how to config s_http_server_opts_?

cpq, 2020-12-24:
Closing this, since the behavior is as expected.

closed 2020-12-24 by cpq