cesanta/mongoose #1105: Websocket Client have a BUG?

shu7734, 2020-03-22:
I changed Example code to this

	nc = mg_connect_ws(&mgr, ev_handler, "ws://118.24.149.132:8080/echo", NULL, "Origin: http://123.net\r\nUser-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.69 Safari/537.36 Edg/81.0.416.34");

I have not any response,I trafic packet like this

0000   52 54 00 77 c9 c0 fe ee 32 07 ea 07 08 00 45 68   RT.w....2.....Eh
0010   01 70 11 d3 40 00 74 06 ec ca 77 7b e2 e4 ac 1b   .p..@.t...w{....
0020   00 07 fe bd 1f 90 e7 0d 1f fd c9 68 d9 7f 50 18   ...........h..P.
0030   02 00 f3 18 00 00 47 45 54 20 2f 65 63 68 6f 20   ......GET /echo 
0040   48 54 54 50 2f 31 2e 31 0d 0a 55 70 67 72 61 64   HTTP/1.1..Upgrad
0050   65 3a 20 77 65 62 73 6f 63 6b 65 74 0d 0a 43 6f   e: websocket..Co
0060   6e 6e 65 63 74 69 6f 6e 3a 20 55 70 67 72 61 64   nnection: Upgrad
0070   65 0d 0a 53 65 63 2d 57 65 62 53 6f 63 6b 65 74   e..Sec-WebSocket
0080   2d 56 65 72 73 69 6f 6e 3a 20 31 33 0d 0a 53 65   -Version: 13..Se
0090   63 2d 57 65 62 53 6f 63 6b 65 74 2d 4b 65 79 3a   c-WebSocket-Key:
00a0   20 4b 51 41 41 41 43 4e 49 41 41 43 2b 47 41 41    KQAAACNIAAC+GAA
00b0   41 68 47 63 41 41 41 3d 3d 0d 0a 48 6f 73 74 3a   AhGcAAA==..Host:
00c0   20 31 31 38 2e 32 34 2e 31 34 39 2e 31 33 32 3a    118.24.149.132:
00d0   38 30 38 30 0d 0a 4f 72 69 67 69 6e 3a 20 68 74   8080..Origin: ht
00e0   74 70 3a 2f 2f 31 32 33 2e 6e 65 74 0d 0a 55 73   tp://123.net..Us
00f0   65 72 2d 41 67 65 6e 74 3a 20 4d 6f 7a 69 6c 6c   er-Agent: Mozill
0100   61 2f 35 2e 30 20 28 57 69 6e 64 6f 77 73 20 4e   a/5.0 (Windows N
0110   54 20 31 30 2e 30 3b 20 57 69 6e 36 34 3b 20 78   T 10.0; Win64; x
0120   36 34 29 20 41 70 70 6c 65 57 65 62 4b 69 74 2f   64) AppleWebKit/
0130   35 33 37 2e 33 36 20 28 4b 48 54 4d 4c 2c 20 6c   537.36 (KHTML, l
0140   69 6b 65 20 47 65 63 6b 6f 29 20 43 68 72 6f 6d   ike Gecko) Chrom
0150   65 2f 38 31 2e 30 2e 34 30 34 34 2e 36 39 20 53   e/81.0.4044.69 S
0160   61 66 61 72 69 2f 35 33 37 2e 33 36 20 45 64 67   afari/537.36 Edg
0170   2f 38 31 2e 30 2e 34 31 36 2e 33 34 0d 0a         /81.0.416.34..

it's not a standard http packet ,is a Bug?

shu7734, 2020-03-23:
maybe bug from Visual Stdio 2019 newest version
 
MG_ENABLE_SSL is 0

but the code run to mongoose.c:3135

shu7734, 2020-03-24:
Find mongoose Bug, where is mongoose.c:10247

  if (extra_headers.len > 0) {
    mg_printf(nc, "%.*s", (int) extra_headers.len, extra_headers.p);
  }
  mg_printf(nc, "\r\n");

  nc->flags |= MG_F_IS_WEBSOCKET;

  mbuf_free(&auth);
}

replace to
  mg_printf(nc, "\r\n\r\n");

cpq, 2020-12-18:
Thanks for reporting.
It is not clear what the issue is - closing this.
If you think there is still an issue, please modify the websocket unit test in `test/unit_test.c` and make it fail.

closed 2020-12-18 by cpq