cesanta/mongoose #1106: Why is the project solution file so out of date and erroneous?

patlecat, 2020-03-22:
The MSVC project file doesn't match the current code and thus produces heaps of errors like mismatching directory paths (see list below). Also the test code has errors and isn't buildable as is! Why do you upload it in such a erroneous state??

-  "error C2065: 'c_str_ne': undeclared identifier" 
- "warning C4013: 'poll_until' undefined; assuming extern returning int"
- "warning C4267: 'function': conversion from 'size_t' to 'int', possible loss of data"
- "warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data"
- "warning C4267: 'function': conversion from 'size_t' to 'int', possible loss of data"
- "warning C4996: 'rmdir': The POSIX name for this item is deprecated. Instead, use the ISO C and C++ conformant name: _rmdir. See online help for details."
- ...and many more

patlecat, 2020-03-31:
Also it's impossible to even compile mongoose if I take only the .c and .h files alone into MSVC2019 and compile it with an example from this repo. The code is seriously broken!
😖

cpq, 2020-12-25:
No project file - no problem :)
The MSVC project file is removed.
Continuous unit test has automatic builds on MSVC1998 and MSVC2017 - see Github actions for this repo.
The vc2017 build has some warnings, those will be cleaned up soon.

closed 2020-12-25 by cpq