cesanta/mongoose #1107: Websocket client connect to wss has  error codes when sending data.

coolzdp, 2020-03-23:
WSS server echo:
E:\demo\websocket\wssServer\Debug>websocketserver.exe
> Starting SSL server on port 8000, cert from server.pem, key from (null)
> Started on port 8000
> 127.0.0.1:8816 ++ joined
> 烫烫0.0.1:8816 135
> ?27.0.0.1:8816
> ?27.0.0.1:8816
> ?27.0.0.1:8816
> ?27.0.0.1:8816
> 烫烫0.0.1:8816 135

WSS client echo:
> E:\demo\websocket\wssClient\Debug>websocketclient.exe
> 9504-- Connected
> 123456
> 
> 
> 
> 
> 123456

html page coneect to wss server work normal.
but not work normal by using  wss client code of mongoose example

coolzdp, 2020-03-23:
not need wss, ws example already has this

coolzdp, 2020-03-23:
websocket client example code has bug?

coolzdp, 2020-03-23:
i find the problem,
the ws client example code has bug,

 while (msg[n - 1] == '\r' || msg[n - 1] == '\n') n--;
      mg_send_websocket_frame(nc, WEBSOCKET_OP_TEXT, msg, n);_
msg not the valid string.

cpq, 2020-12-17:
Closing this.
Please use the latest codebase: https://github.com/cesanta/mongoose/tree/master/examples/websocket-client

closed 2020-12-17 by cpq