cesanta/mongoose #1114: Error in TCP connection (ERR_EMPTY_RESPONSE)

pucmelda, 2020-05-12:
Hi,
I have project with ESP32-WROOM-32. It reads data from the serial line and provides it on the website. It also acts as a TCP / serial line converter on another port. I use Mongoose server for web server functions. Everything works, but somewhere and sometimes an error occurs during communication - after opening a TCP connection, this connection is terminated on the server side - in response, the TCP flag RST is set (read using Wireshark). Chrome will display an ERR_EMPTY_RESPONSE error. More than 100 products are already in use, this problem only appears somewhere. But in some places almost daily. The problem occurs both with Wifi connection and with wired Ethernet connection (it is implemented by LAN8720 circuit). Listing with simple_monitor will not show anything, even in VERBOSE mode. To eliminate the error, disconnect the network connection is enough - restart the wroom or restart the router or unplug and plug in the Ethernet cable. This problem persists over different versions of esp-idf and mongoose - the first version was esp-idf-v2.1.1 and mongoose 6.14, then esp-idf-v3.3 and mongoose 6.14, and now esp-idf-v4.0 and mongoose 6.17. I don't know if to look for the source of the problem in esp-idf or in mongoose.c. Please advise what this may cause or where to look for the source of the error. Thanks

marcelwirtz, 2020-08-31:
Do you use HTTPS or HTTP? Sounds like a problem i have, but on Windows not on EPSs. I figured out that it works if i do without SSL encryption.

cpq, 2020-12-24:
Please use the latest codebase.
It is hard to suggest something specific - the scope is very broad to be constructive.
If the issue is pressing and affects the business, please contact support@cesanta.com

closed 2020-12-24 by cpq