cesanta/mongoose #1118: can websocket client recieve close reason and close code from server?

deetchtoys, 2020-05-19:


cpq, 2020-05-24:
The WS close message is described at https://tools.ietf.org/html/rfc6455#section-5.5.1
Could you elaborate on what do you mean by "close reason" and "close code", please?

deetchtoys, 2020-05-28:
I use websocket_chat_client.c from examples.
When websocket server send close frame, client just run into `MG_EV_CLOSE`. How can i receive close frame in `MG_EV_WEBSOCKET_FRAME`?

cpq, 2020-05-29:
Currently Mongoose does not send a frame message:

```
        case WEBSOCKET_OP_CLOSE:
          mg_ws_close(nc, wsm.data, wsm.size);
          break;
```

Could you elaborate on why `MG_EV_CLOSE` does not suit?

deetchtoys, 2020-06-03:
Our websocket server may send close frame when it occurs error, and close frame contains some descriptions which imply the error, and my websocket client should log it and upload to bigdata platform.

cpq, 2020-12-27:
Please use the latest codebase, which sends `MG_EV_WS_CTL` events: CLOSE, PING, PONG.

closed 2020-12-27 by cpq