cesanta/mongoose #1126: mg_ssl_if_mbed_err() catch more errors

dd1dd1, 2020-06-28:
Please add the following mbedtls errors in mg_ssl_if_mbed_err() - they show up when running the SSLlabs tool. Final score of mongoose6.16 and mbedtls-2.16.6-167-geab4a7a05 as "A". Good work guys!
K.O.

diff --git a/progs/mongoose616.cxx b/progs/mongoose616.cxx
index 004827d2..fb3a3f7f 100644
--- a/progs/mongoose616.cxx
+++ b/progs/mongoose616.cxx
@@ -5269,6 +5269,24 @@ static enum mg_ssl_if_result mg_ssl_if_mbed_err(struct mg_connection *nc,
     LOG(LL_DEBUG, ("%p TLS connection closed by peer", nc));
     nc->flags |= MG_F_CLOSE_IMMEDIATELY;
     res = MG_SSL_OK;
+  } else if (ret == MBEDTLS_ERR_SSL_INVALID_MAC) {
+    nc->flags |= MG_F_CLOSE_IMMEDIATELY;
+    res = MG_SSL_ERROR;
+  } else if (ret == MBEDTLS_ERR_SSL_BAD_HS_PROTOCOL_VERSION) {
+    nc->flags |= MG_F_CLOSE_IMMEDIATELY;
+    res = MG_SSL_ERROR;
+  } else if (ret == MBEDTLS_ERR_SSL_BAD_HS_CLIENT_KEY_EXCHANGE) {
+    nc->flags |= MG_F_CLOSE_IMMEDIATELY;
+    res = MG_SSL_ERROR;
+  } else if (ret == MBEDTLS_ERR_SSL_NO_CIPHER_CHOSEN) {
+    nc->flags |= MG_F_CLOSE_IMMEDIATELY;
+    res = MG_SSL_ERROR;
+  } else if (ret == MBEDTLS_ERR_SSL_NO_USABLE_CIPHERSUITE) {
+    nc->flags |= MG_F_CLOSE_IMMEDIATELY;
+    res = MG_SSL_ERROR;
+  } else if (ret == MBEDTLS_ERR_SSL_BAD_HS_CLIENT_HELLO) {
+    nc->flags |= MG_F_CLOSE_IMMEDIATELY;
+    res = MG_SSL_ERROR;
   } else {
     LOG(LL_ERROR, ("%p mbedTLS error: -0x%04x", nc, -ret));
     nc->flags |= MG_F_CLOSE_IMMEDIATELY;
daqubuntu:midas$

cpq, 2020-12-24:
Could you elaborate on this more please?
What is the purpose of these extra checks, if they are all handled by the `else` branch anyway?

cpq, 2020-12-28:
Closing this - feel free to reopen, in case please give more context.

closed 2020-12-28 by cpq